PySpice-org/PySpice #222: Ngspice 32 support

Naroek56, 2020-05-20:
### Environment (OS, Python version, PySpice version, simulator)
Windows, Python 3.8.0, PySpice 1.4.2, Ngspice 32
### Expected Behaviour
Run circuit configuration and simulation.
### Actual Behaviour
Currently not possible. ngspice.dll cannot be found and even renaming it results during `simulator=circuit.simulator()` in the following error:
> \<cdef source string\>:7: duplicate declaration of struct ngcomplex

Ngspice 32 fixes some caveats that have lead to several simulation crashes and support for it would be greatly appreciated!

FabriceSalvaire, 2020-05-20:
1.4.2 is broken on Windows ... will fix asap

Naroek56, 2020-05-20:
Much appreciated!

FabriceSalvaire, 2020-05-26:
just pushed the code, must make a test release asap ...

you must run now `pyspice-post-installation --install-dll` to download the Windows DLL

Naroek56, 2020-05-28:
Thanks Fabrice, by now I downgraded back to 1.3.2 and will wait for the full release. Let me know if you want me to test it though.

FabriceSalvaire, 2021-05-14:
Outdated and fixed

closed 2021-05-14 by FabriceSalvaire