PySpice-org/PySpice #234: Ubuntu 18.04 Missing libngspice.so steps to install

davehaglan, 2020-06-06:
### Environment (Ubuntu 18.04, Python 3.6, PySpice 1.4.2, ngspice)

### Ubuntu Issue

Running ngspice gives the error:
OSError: cannot load library 'libngspice.so': libngspice.so: cannot open shared object file: No such file or directory. Additionally, ctypes.util.find_library() did not manage to locate a library called 'libngspice.so'

### Steps to Fix it

Several people seem to have trouble running pypice/ngspice on Ubuntu 18.04.  I was able to generate the libngspice.so files following the procedure below.

Install ngspice32 following the instructions in section 32.1.5 of the documentation.

Once the install is complete link the libngspice.so files to the unix library.
cd /usr/lib/x86_64-linux-gnu
sudo ln -s /usr/local/lib/libngspice.so.0.0.0 libngspice.so
sudo ln -s /usr/local/lib/libngspice.so.0.0.0 libngspice.so.0
sudo ln -s /usr/local/lib/libngspice.so.0.0.0 libngspice.so.0.0.0

Install pyspice from git (PIP install was unable to run the pyspice-post-installation --check-install)
git clone https://github.com/FabriceSalvaire/PySpice.git

After this I was able to run the RC example.

Forum post that led to the solution above:
https://forum.kicad.info/t/ngspice-30-simulator-available/14426/2

FabriceSalvaire, 2020-06-06:
These instructions (section 32.1.5) are probably incomplete.

Did you look at https://pyspice.fabrice-salvaire.fr/releases/v1.4/installation.html#ngspice-installation ?  You should be able to install

Ubuntu: libngspice0 https://packages.ubuntu.com/search?suite=default&section=all&arch=any&keywords=libngspice&searchon=names

davehaglan, 2020-06-07:
I was not able to install the libngspice.so packages from the Ubuntu page.  The Ubuntu packages are for Ubuntu versions newer than 18.04.

I had installed ngspice using "apt-get install ngspice" but it did not install the libngspice.so and I was not able to run any examples due to the missing libngspice.so.  After, I followed the steps above I was able to run the examples successfully.  I posted my method to install the libngspice.so since it seemed like a couple of Ubuntu 18.04 users had the same problem.

Still learning the ngspice package but I like what I see so far.  Thanks!

FabriceSalvaire, 2020-06-11:
@davehaglan I cannot reproduce this issue, these links should have been created by make install