PySpice-org/PySpice #240: documentation build error

efferre79, 2020-06-29:
### Environment (OS, Python version, PySpice version, simulator)
Gentoo Linux, python3.6, PySpice 1.4.2, ngspice

### Expected Behaviour
build process should complete correctly

### Actual Behaviour
build error:

```
make -j8 -l8 -f Makefile.sphinx html 
sphinx-build -b html -d build/doctrees   source build/html
Running Sphinx v2.4.4

Configuration error:
There is a programmable error in your configuration file:

Traceback (most recent call last):
  File "/usr/lib64/python3.6/site-packages/sphinx/config.py", line 348, in eval_config_file
    execfile_(filename, namespace)
  File "/usr/lib64/python3.6/site-packages/sphinx/util/pycompat.py", line 81, in execfile_
    exec(code, _globals)
  File "/var/tmp/portage/sci-electronics/PySpice-1.4.2/work/PySpice-1.4.2/doc/sphinx/source/conf.py", line 100, in <module>
    release = setup_dict['version']
KeyError: 'version'

make: *** [Makefile.sphinx:42: html] Error 2
```

### Steps to reproduce the behaviour
python3.6 setup.py build

FabriceSalvaire, 2020-07-04:
could you try v1.4.3 to be sure it is not fixed

efferre79, 2020-07-07:
tested with v1.4.3 and it works, closing this issue then, thanks

closed 2020-07-07 by efferre79