PySpice-org/PySpice #247: Wrong target in hyperlink for FAQ in README.rst

cyber-g, 2020-08-03:
### Environment (OS, Python version, PySpice version, simulator)
Online with a regular web browser

### Steps to reproduce the behaviour
Load the PySpice project page :
https://github.com/FabriceSalvaire/PySpice
or the README file : 
https://github.com/FabriceSalvaire/PySpice/blob/master/README.rst

And click on the hyperlink of the FAQ : 
> Implemented raw spice pass through (see [User FAQ](https://github.com/FabriceSalvaire/PySpice/blob/master/faq.html))

### Expected Behaviour
The User FAQ should be loaded

### Actual Behaviour
Error 404. 

### Solution
The right target is : 
https://github.com/FabriceSalvaire/PySpice/blob/master/doc/sphinx/source/faq.rst

(I will submit shortly a Pull request with the solution)

cyber-g, 2020-08-03:
I dug a little bit more and figured out that target links are not pointing to github architecture directory but are designed for the documentation website (https://pyspice.fabrice-salvaire.fr)

That being said, the link in question does not work on the documentation site either. 
> Implemented raw spice pass through (see [User FAQ](https://pyspice.fabrice-salvaire.fr/faq.html))

(found at [News](https://pyspice.fabrice-salvaire.fr/releases/v1.4/news.html) )

Since, I am not sure about the right approach to fix the issue, _I will not finally propose a pull request_. 
I would actually need some guidance

FabriceSalvaire, 2021-05-07:
Should be due to the fact news.rst is used to generate the doc and README which are hosted on two different hosts ...