PySpice-org/PySpice #248: OSError: cannot load library ngspice.dll': error 0x7e

HR42, 2020-08-03:
### Environment (OS, Python version, PySpice version, simulator)
Win10.0.19041
Python 3.8.5 (tags/v3.8.5:580fbb0, Jul 20 2020, 15:57:54) [MSC v.1924 64 bit (AMD64)] on win32 
PySpice 1.4.3
NgSpice

### Expected Behaviour
The ngspice.dll should load without an error.

### Actual Behaviour
Traceback (most recent call last):
  File "test_pyspice.py", line 16, in <module>
    simulator = circuit.simulator(temperature=25, nominal_temperature=25)
  File "C:\Python38\lib\site-packages\PySpice\Spice\Netlist.py", line 1299, in simulator
    return CircuitSimulator.factory(self, *args, **kwargs)
  File "C:\Python38\lib\site-packages\PySpice\Spice\Simulation.py", line 1119, in factory
    return sub_cls(circuit, *args, **kwargs)
  File "C:\Python38\lib\site-packages\PySpice\Spice\NgSpice\Simulation.py", line 99, in __init__
    self._ngspice_shared = NgSpiceShared.new_instance()
  File "C:\Python38\lib\site-packages\PySpice\Spice\NgSpice\Shared.py", line 452, in new_instance
    instance = cls(ngspice_id=ngspice_id, send_data=send_data, verbose=verbose)
  File "C:\Python38\lib\site-packages\PySpice\Spice\NgSpice\Shared.py", line 480, in __init__
    self._load_library(verbose)
  File "C:\Python38\lib\site-packages\PySpice\Spice\NgSpice\Shared.py", line 540, in _load_library
    self._ngspice_shared = ffi.dlopen(self.library_path)
  File "C:\Python38\lib\site-packages\cffi\api.py", line 150, in dlopen
    lib, function_cache = _make_ffi_library(self, name, flags)
  File "C:\Python38\lib\site-packages\cffi\api.py", line 832, in _make_ffi_library
    backendlib = _load_backend_lib(backend, libname, flags)
  File "C:\Python38\lib\site-packages\cffi\api.py", line 828, in _load_backend_lib
    return backend.load_library(path, flags)
OSError: cannot load library 'C:\Python38\lib\site-packages\PySpice\Spice\NgSpice\Spice64_dll\dll-vs\ngspice.dll': error 0x7e

### Steps to reproduce the behaviour
I try to run the "Resistor Bridge" example on a fresh installation.
I have installed pyspice direct with "pip install pyspice".
Afterwards the I have run "pyspice-post-installation --install-ngspice-dll".
The installation steps have processed without any error.
The test "pyspice-post-installation --check-install" gives the same error as shown above.

Do you have any hint how to solve this problem?
I want to avoid Anaconda, hence the direct installation with pip.
As far as I understand the docs this should also be possible?

Regards HR42

HR42, 2020-08-04:
Oh, I found the solution: The NgSpice DLL folder must be on the windows path.
Fortunately this can be made within the python script:
In my case it is: 
```
import os
os.environ["path"] += os.pathsep + r"C:\Python38\lib\site-packages\PySpice\Spice\NgSpice\Spice64_dll\dll-vs"
```
With this modification the example works as expected.

A specific hint might be helpful when this particular exception occurs?
E.g. "Please ensure that the DLL folder is on the windows PATH."

Regards HR42

OK83, 2022-09-01:
For me I solved solved this with followed steps :
  1. install visual studio code
  2. create venv
  3. install al needed python libs in venv (numpy, matplotlib, pyspice(don't know if it is needed))
  4. change intepreter in visual studio code
  5. in venv run : pyspice-post-installation --install-ngspice-dll
  6. in venv run : pyspice-post-installation --check-install ->  PySpice should work as expected

sebrupi, 2024-03-08:
For me i solved it with:

[image]

run this over the Terminal:
1) cd in to  -->   C:\Users\**NAME**\AppData\Roaming\Python\Python39\Scripts

2) run this comand --> .\pyspice-post-installation  --install-ngspice-dll

sebrupi, 2024-03-08:
For me i solved it with:

[image]

run this over the Terminal:

cd in to  --> cd C:\Users\**NAME**\AppData\Roaming\Python\Python39\Scripts

run this comand --> .\pyspice-post-installation --install-ngspice-dll

sebrupi, 2024-03-08:
For me i solved it with:

[image]

run this 2 comands over the Terminal:

1) cd in to --> cd C:\Users*NAME*\AppData\Roaming\Python\Python39\Scripts

2) run this comand --> .\pyspice-post-installation --install-ngspice-dll

sebrupi, 2024-03-08:
For me i solved it with:

[image]

run this 2 comands over the Terminal:

1) cd in to --> cd C:\UsersNAME\AppData\Roaming\Python\Python39\Scripts

2) run this comand --> .\pyspice-post-installation --install-ngspice-dll

sebrupi, 2024-03-08:
For me i solved it with:

[image]

run this 2 comands over the Terminal:

1.   cd in to --> cd C:\Users\YOUR_PC_NAME\AppData\Roaming\Python\Python39\Scripts

2.   run this comand --> .\pyspice-post-installation --install-ngspice-dll