PySpice-org/PySpice #255: Unable to use external current source with ngspice_shared

krillin666, 2020-09-08:
Hello, I am trying to plug an external current source into a circuit I made with Pyspice. I first wanted to test the one in this [example](https://pyspice.fabrice-salvaire.fr/releases/v1.4/examples/ngspice-shared/external-source.html?highlight=voltage%20source). I modified the code such that the plug source was a current one and not a voltage. However I haven't had any success because every attempt I make to solve I get the error **"parameter value out of range or the wrong type"**.

I thought that maybe the current values were not being generated properly and as such were causing the given error. However, when I added print statements to the MyNgSpiceShared class I noticed (see output bellow) that the get_isrc_data function is not even called.

## Output

> BEFORE FUNCTIONS
Inside __init__
[1;32m2020-09-08 17:22:13,453[0m - [1;34mPySpice.Spice.NgSpice.Shared.NgSpiceShared.load_circuit[0m - [1;31mERROR[0m - 
Circuit: Current Divider
Error on line 4 :
iinput 1 0 dc 0 external
parameter value out of range or the wrong type
Traceback (most recent call last):

## The code I am using is bellow: 

    import math

    import matplotlib.pyplot as plt

    import PySpice.Logging.Logging as Logging
    logger = Logging.setup_logging()
    import locale
    locale.setlocale(locale.LC_NUMERIC, 'C')
    from PySpice.Probe.Plot import plot
    from PySpice.Spice.Netlist import Circuit
    from PySpice.Spice.NgSpice.Shared import NgSpiceShared
    from PySpice.Unit import *

        

    class MyNgSpiceShared(NgSpiceShared):
        print("BEFORE FUNCTIONS")

        def __init__(self, amplitude, frequency, **kwargs):

            super().__init__(**kwargs)

            self._amplitude = amplitude
            self._pulsation = float(frequency.pulsation)
            print("Inside __init__")

        def get_vsrc_data(self, voltage, time, node, ngspice_id):
            self._logger.debug('ngspice_id-{} get_vsrc_data @{} node {}'.format(ngspice_id, time, node))
            print("INSIDE get_vsrc_data")
            return 0

        def get_isrc_data(self, current, time, node, ngspice_id):

            print("INSIDE get_isrc_data")
            self._logger.debug('ngspice_id-{} get_isrc_data @{} node {}'.format(ngspice_id, time, node))
            current[0] = self._amplitude * math.sin(self._pulsation * time)

            print(current)
            return 0

    circuit = Circuit('Current Divider')

    amplitude = u_A(10)
    frequency = 50@u_Hz
    circuit.R(1, 1, circuit.gnd, 2@u_kΩ)
    circuit.R(2, 1, circuit.gnd, 1@u_kΩ)

    circuit.I('input', 1, circuit.gnd, "dc 0 external") # Fixme: current value
    for resistance in (circuit.R1, circuit.R2):
        resistance.minus.add_current_probe(circuit) # to get positive value

    ngspice_shared = MyNgSpiceShared(amplitude=amplitude, frequency=frequency, send_data=False)
    simulator = circuit.simulator(temperature=25, nominal_temperature=25,
                                  simulator='ngspice-shared', ngspice_shared=ngspice_shared)
    period = float(frequency.period)
    analysis = simulator.transient(step_time=period/200, end_time=period*2)

Thank you for your time.

thejackal360, 2022-01-12:
This appears to be an issue with ngspice. External current sources don't appear to work.

https://github.com/thejackal360/ngspice-bug-reproduction

thejackal360, 2022-01-17:
This appears to be an ngspice issue. I fixed the bug on SourceForge and made a merge request:

https://sourceforge.net/p/ngspice/ngspice/merge-requests/13/

Try this fix. Let me know if you need any help compiling libngspice.so from source.

justincui, 2023-02-06:
Issue: cannot run this example in windows 10 : https://pyspice.fabrice-salvaire.fr/releases/v1.5/examples/ngspice-shared/external-source.html?highlight=external#
###################### environment (windows10, anaconda) #######################
>python --version
Python 3.9.16

>conda list | grep spice
# packages in environment at C:\Users\justin.cui\Anaconda3\envs\spice:
ngspice                   38                   h57928b3_0    conda-forge
ngspice-exe               38                   h295a5eb_0    conda-forge
ngspice-lib               38                   h295a5eb_0    conda-forge
pyspice                   1.5              py39hcbf5309_1    conda-forge
###############################To reproduce #####################################
1. In ipython, run the python code of the example directly. It just prints the following output and then ipython quits silently:
`2023-02-06 12:03:37,546 - PySpice.Spice.NgSpice.Shared.NgSpiceShared - Shared.WARNING - Unsupported Ngspice version 38`
2. add logger.setLevel('DEBUG') and run the code again. It prints the following and ipython quits silently:
`2023-02-06 12:08:11,677 - PySpice.Spice.Netlist - Netlist.DEBUG - Element don't define a PINS attribute
2023-02-06 12:08:11,677 - PySpice.Spice.Netlist - Netlist.DEBUG - FixedPinElement don't define a PINS attribute
2023-02-06 12:08:11,679 - PySpice.Spice.Netlist - Netlist.DEBUG - SubCircuitElement don't define a PINS attribute
2023-02-06 12:08:11,685 - PySpice.Spice.Netlist - Netlist.DEBUG - Resistor don't define a PINS attribute
2023-02-06 12:08:11,686 - PySpice.Spice.Netlist - Netlist.DEBUG - SemiconductorResistor don't define a PINS attribute
2023-02-06 12:08:11,686 - PySpice.Spice.Netlist - Netlist.DEBUG - BehavioralResistor don't define a PINS attribute
2023-02-06 12:08:11,687 - PySpice.Spice.Netlist - Netlist.DEBUG - Capacitor don't define a PINS attribute
2023-02-06 12:08:11,687 - PySpice.Spice.Netlist - Netlist.DEBUG - SemiconductorCapacitor don't define a PINS attribute
2023-02-06 12:08:11,687 - PySpice.Spice.Netlist - Netlist.DEBUG - BehavioralCapacitor don't define a PINS attribute
2023-02-06 12:08:11,688 - PySpice.Spice.Netlist - Netlist.DEBUG - Inductor don't define a PINS attribute
2023-02-06 12:08:11,688 - PySpice.Spice.Netlist - Netlist.DEBUG - BehavioralInductor don't define a PINS attribute
2023-02-06 12:08:11,689 - PySpice.Spice.Netlist - Netlist.DEBUG - CoupledInductor don't define a PINS attribute
2023-02-06 12:08:11,689 - PySpice.Spice.Netlist - Netlist.DEBUG - VoltageControlledSwitch don't define a PINS attribute
2023-02-06 12:08:11,689 - PySpice.Spice.Netlist - Netlist.DEBUG - CurrentControlledSwitch don't define a PINS attribute
2023-02-06 12:08:11,690 - PySpice.Spice.Netlist - Netlist.DEBUG - VoltageSource don't define a PINS attribute
2023-02-06 12:08:11,690 - PySpice.Spice.Netlist - Netlist.DEBUG - CurrentSource don't define a PINS attribute
2023-02-06 12:08:11,691 - PySpice.Spice.Netlist - Netlist.DEBUG - VoltageControlledCurrentSource don't define a PINS attribute
2023-02-06 12:08:11,691 - PySpice.Spice.Netlist - Netlist.DEBUG - VoltageControlledVoltageSource don't define a PINS attribute
2023-02-06 12:08:11,691 - PySpice.Spice.Netlist - Netlist.DEBUG - CurrentControlledCurrentSource don't define a PINS attribute
2023-02-06 12:08:11,692 - PySpice.Spice.Netlist - Netlist.DEBUG - CurrentControlledVoltageSource don't define a PINS attribute
2023-02-06 12:08:11,692 - PySpice.Spice.Netlist - Netlist.DEBUG - BehavioralSource don't define a PINS attribute
2023-02-06 12:08:11,692 - PySpice.Spice.Netlist - Netlist.DEBUG - NonLinearVoltageSource don't define a PINS attribute
2023-02-06 12:08:11,693 - PySpice.Spice.Netlist - Netlist.DEBUG - NonLinearCurrentSource don't define a PINS attribute
2023-02-06 12:08:11,694 - PySpice.Spice.Netlist - Netlist.DEBUG - JunctionFieldEffectTransistor don't define a PINS attribute
2023-02-06 12:08:11,694 - PySpice.Spice.Netlist - Netlist.DEBUG - Mesfet don't define a PINS attribute
2023-02-06 12:08:11,694 - PySpice.Spice.Netlist - Netlist.DEBUG - LosslessTransmissionLine don't define a PINS attribute
2023-02-06 12:08:11,695 - PySpice.Spice.Netlist - Netlist.DEBUG - LossyTransmission don't define a PINS attribute
2023-02-06 12:08:11,695 - PySpice.Spice.Netlist - Netlist.DEBUG - CoupledMulticonductorLine don't define a PINS attribute
2023-02-06 12:08:11,696 - PySpice.Spice.Netlist - Netlist.DEBUG - SingleLossyTransmissionLine don't define a PINS attribute
2023-02-06 12:08:11,696 - PySpice.Spice.Netlist - Netlist.DEBUG - XSpiceElement don't define a PINS attribute
2023-02-06 12:08:11,696 - PySpice.Spice.Netlist - Netlist.DEBUG - GSSElement don't define a PINS attribute
2023-02-06 12:08:11,706 - PySpice.Spice.Netlist - Netlist.DEBUG - SinusoidalVoltageSource don't define a PINS attribute
2023-02-06 12:08:11,706 - PySpice.Spice.Netlist - Netlist.DEBUG - SinusoidalCurrentSource don't define a PINS attribute
2023-02-06 12:08:11,707 - PySpice.Spice.Netlist - Netlist.DEBUG - AcLine don't define a PINS attribute
2023-02-06 12:08:11,708 - PySpice.Spice.Netlist - Netlist.DEBUG - PulseVoltageSource don't define a PINS attribute
2023-02-06 12:08:11,708 - PySpice.Spice.Netlist - Netlist.DEBUG - PulseCurrentSource don't define a PINS attribute
2023-02-06 12:08:11,709 - PySpice.Spice.Netlist - Netlist.DEBUG - ExponentialVoltageSource don't define a PINS attribute
[... cut ...]
.tran 0.0001s 0.04s 0s
.end

2023-02-06 12:08:11,910 - PySpice.Spice.NgSpice.Shared - Shared.DEBUG - b'stdout Note: No compatibility mode selected!'
2023-02-06 12:08:11,910 - PySpice.Spice.NgSpice.Shared - Shared.DEBUG - b'stdout Circuit: Voltage Divider'
2023-02-06 12:08:11,912 - PySpice.Spice.NgSpice.Shared.NgSpiceShared - Shared.DEBUG - Execute command: run
2023-02-06 12:08:11,913 - PySpice.Spice.NgSpice.Shared - Shared.DEBUG - b'stdout Doing analysis at TEMP = 25.000000 and TNOM = 25.000000'`