PySpice-org/PySpice #264: Error: there aren't any circuits loaded.

pepijndevos, 2020-11-17:
### Environment (OS, Python version, PySpice version, simulator)

Arch Linux, Python 3.8.6, PySpice 1.4.3 (and git master/devel), Ngspice 33.

### Expected Behaviour

Post installation check completes

### Actual Behaviour

Post installation check, and any examples I run give the following error.

### Steps to reproduce the behaviour

On Arch Linux with Ngspice 33

* Checkout latest git version
* apply #262  (or not, same behaviour)
* `pip install .`
* run self check

```
$ pyspice-post-installation --check-install
OS: linux

Environments:
PATH /home/pepijn/code/asic/pschem/env/bin:/usr/local/bin:/usr/local/sbin:/usr/bin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl
LD_LIBRARY_PATH undefined
PYTHONPATH undefined
NGSPICE_LIBRARY_PATH undefined
SPICE_LIB_DIR undefined
SPICE_EXEC_DIR undefined
SPICE_ASCIIRAWFILE undefined
SPICE_SCRIPTS undefined
NGSPICE_MEAS_PRECISION undefined
SPICE_NO_DATASEG_CHECK undefined
NGSPICE_INPUT_DIR undefined

On Virtual Environment:
VIRTUAL_ENV /home/pepijn/code/asic/pschem/env

Load PySpice module
loaded /home/pepijn/code/asic/pschem/env/lib/python3.8/site-packages/PySpice/__init__.py version 1.4.3

ngspice supported version: 33

NgSpiceShared configuration is
  NgSpiceShared.NGSPICE_PATH = None
  NgSpiceShared.LIBRARY_PATH = libngspice{}.so

Working directory: /home/pepijn/code/asic/PySpice

NgSpice: /home/pepijn/code/asic/pschem/env/lib/python3.8/site-packages/PySpice/Spice/NgSpice
/home/pepijn/code/asic/pschem/env/lib/python3.8/site-packages/PySpice/Spice/NgSpice SimulationType.py
/home/pepijn/code/asic/pschem/env/lib/python3.8/site-packages/PySpice/Spice/NgSpice Simulation.py
/home/pepijn/code/asic/pschem/env/lib/python3.8/site-packages/PySpice/Spice/NgSpice RawFile.py
/home/pepijn/code/asic/pschem/env/lib/python3.8/site-packages/PySpice/Spice/NgSpice Server.py
/home/pepijn/code/asic/pschem/env/lib/python3.8/site-packages/PySpice/Spice/NgSpice api.h
/home/pepijn/code/asic/pschem/env/lib/python3.8/site-packages/PySpice/Spice/NgSpice __init__.py
/home/pepijn/code/asic/pschem/env/lib/python3.8/site-packages/PySpice/Spice/NgSpice Shared.py
/home/pepijn/code/asic/pschem/env/lib/python3.8/site-packages/PySpice/Spice/NgSpice/__pycache__ Server.cpython-38.pyc
/home/pepijn/code/asic/pschem/env/lib/python3.8/site-packages/PySpice/Spice/NgSpice/__pycache__ __init__.cpython-38.pyc
/home/pepijn/code/asic/pschem/env/lib/python3.8/site-packages/PySpice/Spice/NgSpice/__pycache__ SimulationType.cpython-38.pyc
/home/pepijn/code/asic/pschem/env/lib/python3.8/site-packages/PySpice/Spice/NgSpice/__pycache__ Simulation.cpython-38.pyc
/home/pepijn/code/asic/pschem/env/lib/python3.8/site-packages/PySpice/Spice/NgSpice/__pycache__ Shared.cpython-38.pyc
/home/pepijn/code/asic/pschem/env/lib/python3.8/site-packages/PySpice/Spice/NgSpice/__pycache__ RawFile.cpython-38.pyc

OS is Linux
Found in library search path: libngspice.so.0

Load NgSpiceShared
Load library libngspice.so
loaded /usr/lib/libngspice.so.0.0.0

Ngspice version is 33
  has xspice: True
  has cider True

> version -f
******
** ngspice-33 : Circuit level simulation program
** The U. C. Berkeley CAD Group
** Copyright 1985-1994, Regents of the University of California.
** Copyright 2001-2020, The ngspice team.
** Please get your ngspice manual from http://ngspice.sourceforge.net/docs.html
** Please file your bug-reports at http://ngspice.sourceforge.net/bugrep.html
** Creation Date: Thu Oct 22 18:23:04 UTC 2020
**
** CIDER 1.b1 (CODECS simulator) included
** XSPICE extensions included
** Relevant compilation options (refer to user's manual):
** OpenMP multithreading for BSIM3, BSIM4 enabled
** X11 interface not compiled into ngspice
**
******

2020-11-17 19:52:12,315 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Error: there aren't any circuits loaded.
Traceback (most recent call last):
  File "/home/pepijn/code/asic/pschem/env/bin/pyspice-post-installation", line 8, in <module>
    sys.exit(main())
  File "/home/pepijn/code/asic/pschem/env/lib/python3.8/site-packages/PySpice/Scripts/pyspice_post_installation.py", line 452, in main
    return _.run()
  File "/home/pepijn/code/asic/pschem/env/lib/python3.8/site-packages/PySpice/Scripts/pyspice_post_installation.py", line 164, in run
    self.check_installation()
  File "/home/pepijn/code/asic/pschem/env/lib/python3.8/site-packages/PySpice/Scripts/pyspice_post_installation.py", line 418, in check_installation
    circuit_test.test_spinit()
  File "/home/pepijn/code/asic/pschem/env/lib/python3.8/site-packages/PySpice/Scripts/pyspice_post_installation.py", line 98, in test_spinit
    analysis = simulator.transient(step_time=1@U.u_us, end_time=500@U.u_us)
  File "/home/pepijn/code/asic/pschem/env/lib/python3.8/site-packages/PySpice/Spice/Simulation.py", line 1161, in transient
    return self._run('transient', *args, **kwargs)
  File "/home/pepijn/code/asic/pschem/env/lib/python3.8/site-packages/PySpice/Spice/NgSpice/Simulation.py", line 118, in _run
    self._ngspice_shared.run()
  File "/home/pepijn/code/asic/pschem/env/lib/python3.8/site-packages/PySpice/Spice/NgSpice/Shared.py", line 1168, in run
    self.exec_command(command)
  File "/home/pepijn/code/asic/pschem/env/lib/python3.8/site-packages/PySpice/Spice/NgSpice/Shared.py", line 842, in exec_command
    raise NgSpiceCommandError("Command '{}' failed".format(command))
PySpice.Spice.NgSpice.Shared.NgSpiceCommandError: Command 'run' failed
```

diadatp, 2020-11-25:
I'm facing the same issue. Same log output on doing the post install check. However using `NgSpiceShared.new_instance()` to create and manage simulations allows me to successfully run my own circuits.

pepijndevos, 2020-11-25:
I think I may actually know what is causing this. I'm writing my own wrapper for ngSpice now, and had the same problem. It turns out that ngSpice expects an empty newline at the end of the circuit. So if you don't supply one it chops of you `.end` line and never parses the circuit. Not sure if this is a bug as there is an explicit comment in the source that documents this, but it does contradict the examples in the manual and working code out there. So from the PySpice side the workaround is to append a newline at the end of the circuit, but maybe this should be fixed upstream. I have signed up for their mailing list to ask, but have not yet be admitted I think.

benedictjones, 2020-12-14:
I don't think NGSpice 33 is supported yet. I was helping someone set up PySpice on Mac and he was getting errors when checking the install. Everything worked when he downgraded to ngspice 31/32

cyber-g, 2021-01-25:
I confirm that Ngspice 33 is not supported yet. 
The attached [Dockerfile](https://github.com/FabriceSalvaire/PySpice/files/5869844/DockerfileUbuntuHirsute.zip) illustrates the issue. It "loads" Ubuntu Hirsute which includes libngspice0 version 33. 
```
2021-01-25 21:38:34,913 - PySpice.Doc.ExampleTools.find_libraries - INFO - SPICE library path is /root/PySpice/examples/libraries
2021-01-25 21:38:35,067 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._init_ngspice - WARNING - Unsupported Ngspice version 33
2021-01-25 21:38:35,071 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Error: there aren't any circuits loaded.
Traceback (most recent call last):
  File "/root/PySpice/examples/transistor/transistor.py", line 56, in <module>
    analysis = simulator.dc(Vbase=slice(0, 3, .01))
  File "/usr/local/lib/python3.9/dist-packages/PySpice/Spice/Simulation.py", line 1151, in dc
    return self._run('dc', *args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/PySpice/Spice/NgSpice/Simulation.py", line 118, in _run
    self._ngspice_shared.run()
  File "/usr/local/lib/python3.9/dist-packages/PySpice/Spice/NgSpice/Shared.py", line 1168, in run
    self.exec_command(command)
  File "/usr/local/lib/python3.9/dist-packages/PySpice/Spice/NgSpice/Shared.py", line 842, in exec_command
    raise NgSpiceCommandError("Command '{}' failed".format(command))
PySpice.Spice.NgSpice.Shared.NgSpiceCommandError: Command 'run' failed
access control enabled, only authorized clients can connect
```
Change "hirsute" to " groovy", which includes libngspice0 version 32.2, and there is no error. 

Here is a [Bash script](https://github.com/FabriceSalvaire/PySpice/files/5869895/launchUbuntuPySpiceDocker.zip) to launch the docker image on Linux.

closed 2021-05-08 by FabriceSalvaire