PySpice-org/PySpice #266: Needs requests as dependency?

endolith, 2020-12-16:
### Environment (OS, Python version, PySpice version, simulator)

Windows 10 64-bit, conda 4.9.2, python 3.9.1

### Expected Behaviour

`pyspice-post-installation --install-ngspice-dll` should install the DLL

### Actual Behaviour

It complained that `requests` was not installed

### Steps to reproduce the behaviour

I ran https://pyspice.fabrice-salvaire.fr/releases/v1.4/installation.html#on-windows

I created a blank conda environment, activated it, ran `conda install -c conda-forge pyspice`, then ran `pyspice-post-installation --install-ngspice-dll`

umesh-timalsina, 2020-12-18:
I can attest to same for Linux.

fdion, 2021-01-15:
Same issue on linux using conda, post install it errors out doing `pyspice-post-installation --check-install`
```
ModuleNotFoundError: No module named 'requests'
```

Workaround is to `conda install requests`

FabriceSalvaire, 2021-05-07:
Fixed in PR #270

commit 6eea7ddd (2021-05-07): Add dependency for requests library

Fixes #266
closed 2021-05-07 by FabriceSalvaire