PySpice-org/PySpice #282: `pyspice-post-installation` script fails to download ngspice from SourceForge.

gbmhunter, 2021-05-09:
### Environment (OS, Python version, PySpice version, simulator)

OS: Windows 10
Python Version: 3.8.8
PySpice Version: 1.4.3

### Expected Behaviour

ngspice installation should succeed when using the `pyspice-post-installation` script. For example, calling `pyspice-post-installation --install-ngspice-dll --ngspice-version=32` should work.

### Actual Behaviour

Installation fails due to `ngspice` files not being found on SourceForge . This is due to the owners of ngspice on SourceForge placing the files under an sub-directory of `ng-spice-rework` called `old-releases` for all versions of ngspice up to and including 32, see https://sourceforge.net/projects/ngspice/files/ng-spice-rework/old-releases/. 

### Steps to reproduce the behaviour

`pyspice-post-installation --install-ngspice-dll --ngspice-version=32`

@FabriceSalvaire I'm happy to attempt to make a fix and create a pull request for you. I think there just needs to be some logic to change the `NGSPICE_RELEASE_URL` depending on the version.

FabriceSalvaire, 2021-05-10:
Do you need an old release ? This is now fixed for the latest.

gbmhunter, 2021-05-11:
Thanks, I was trying an older release because I was getting a warning that v33 or v34 were too new and not supported.

closed 2021-05-11 by gbmhunter