PySpice-org/PySpice #291: SpiceParser.build_circuit() does not build subcircuits

jimzenn, 2021-05-16:
### Environment (OS, Python version, PySpice version, simulator)

OS: Linux zealoft 5.11.0-1006-gcp #6-Ubuntu SMP Wed Apr 14 21:09:18 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
Python version: 3.9.4
PySpice version: 1.4.3
Simulator: ngSpice

### Related Code

```
class ParallelResistor2(SubCircuit):
    __nodes__ = ('n1', 'n2')
    def __init__(self, name, R1=1@u_Ω, R2=2@u_Ω):
        SubCircuit.__init__(self, name, *self.__nodes__)
        self.R(1, 'n1', 'n2', R1)
        self.R(2, 'n1', 'n2', R2)

circuit = Circuit('Test')
circuit.subcircuit(ParallelResistor2('pr1', R2=2@u_Ω))
circuit.X('1', 'pr1', 1, circuit.gnd)
circuit.subcircuit(ParallelResistor2('pr2', R2=3@u_Ω))
circuit.X('2', 'pr2', 1, circuit.gnd)

source = str(circuit)

parser = SpiceParser(source=source)
bootstrap_circuit = parser.build_circuit()
bootstrap_source = str(bootstrap_circuit)

print(bootstrap_source)

assert bootstrap_source == source

```

### Expected Behaviour

```
.title Test
.subckt pr1 n1 n2
R1 n1 n2 1Ohm
R2 n1 n2 2Ohm
.ends pr1

.subckt pr2 n1 n2
R1 n1 n2 1Ohm
R2 n1 n2 3Ohm
.ends pr2
X1 1 0 pr1
X2 1 0 pr2
```

### Actual Behaviour

```
.title Test
X1 1 0 pr1
X2 1 0 pr2
```

It seems this feature is not implemented?

https://github.com/FabriceSalvaire/PySpice/blob/e459acbad6adac20d73a82dce324fb94efefa1ce/PySpice/Spice/Parser_jmgc.py#L1440

dontillman, 2023-04-10:
Hi.

I'm having the same problem; the build_circuit() operation is completely ignoring subcircuits.

I think I see what's going on... It appears to be testing the wrong class of subcircuit statements.

You can patch it in place and get work done:

```
diff PySpice/Spice/Parser.py~ PySpice/Spice/Parser.py
1008c1008
<             elif isinstance(statement, SubCircuit):
---
>             elif isinstance(statement, SubCircuitStatement):
```

I'm unfamiliar with PySpice development customs and protocols, so it would be great if someone could patch this in for us.   Thanks!

  -- Don

0range0range, 2025-04-14:
Hello, 

Still not clear how to do it. Latest commit on devel also does not allow old way of import: 

`from PySpice.Spice.Parser import SpiceParser # does not work`

`from PySpice.Spice.Parser.Parser import SpiceParser #  works`

I am not sure if second way is a desired way.

Kreijstal, 2025-04-14:
ahh spiceparser is probably kinda shit

tapegoji, 2025-04-14:
Checkout my development branch of PySpice. It can handle many cases. Even skywater pdk can be run with pyspice now. 

I am not sure in subcircuit but I can fix it if still is an issue. Try it and let me know

tapegoji, 2025-04-15:
really? only some of the examples, I thought. Is there any specific one that is causing issue?

Kreijstal, 2025-04-15:
also consider opening issues on your fork

0range0range, 2025-04-16:
Hi @tapegoji 
> Checkout my development branch of PySpice. It can handle many cases. Even skywater pdk can be run with pyspice now.
> 
> I am not sure in subcircuit but I can fix it if still is an issue. Try it and let me know

I tried on your development branch, but the issue persists. It is the same problem as in original post. Instances of subcirucits are being parsed, but not the actual subcircuit definition.

General question is: what is the state of the project? I see that your development pull request already diverged from master. Will it be developed further?

tapegoji, 2025-04-16:
Fabrice, seems to not have enough time to work on it. He might be working on it. I am not sure. Or the project is left abandoned.  it might be dead. I plan to work on this and fix the bugs for my own needs. 

send me a simple example and I will fix the issue.

tapegoji, 2025-04-16:
There is a temp fix for it now. look into the example/spice-parser/parse-subcircuit-example.py

`
import PySpice.Logging.Logging as Logging
logger = Logging.setup_logging()

####################################################################################################

from PySpice import Circuit, SubCircuit, SubCircuitFactory
from PySpice.Spice.Parser.HighLevelParser import SpiceSource
from PySpice.Spice.Parser import Translator
from PySpice.Unit import *

class ParallelResistor2(SubCircuit):
    __nodes__ = ('n1', 'n2')
    def __init__(self, name, R1=1@u_Ω, R2=2@u_Ω):
        SubCircuit.__init__(self, name, *self.__nodes__)
        self.R(1, 'n1', 'n2', R1)
        self.R(2, 'n1', 'n2', R2)

circuit = Circuit('Test')
circuit.R('1', 'input', 'n1', 1@u_Ω)
circuit.subcircuit(ParallelResistor2('pr1', R2=2@u_Ω))
circuit.X('1', 'pr1', 1, circuit.gnd)
circuit.subcircuit(ParallelResistor2('pr2', R2=3@u_Ω))
circuit.X('2', 'pr2', 1, circuit.gnd)

source = str(circuit)
print(circuit)

spice_source = SpiceSource(source=source, title_line=False)
bootstrap_circuit = Translator.Builder().translate(spice_source)
bootstrap_source = str(bootstrap_circuit)

print(bootstrap_source)

assert bootstrap_source == source
`

Kreijstal, 2025-04-17:
@tapegoji if I understand this corerectly. you conver this to spice, then convert it back to circuit, this is undesirable, I'd like to use the pyspice api as it was, without going through this hacks. Otherwise I'd just use spice without pyspice...

0range0range, 2025-04-17:
> There is a temp fix for it now. look into the example/spice-parser/parse-subcircuit-example.py
>
> ` import PySpice.Logging.Logging as Logging logger = Logging.setup_logging() [...]

Hi, 

Thanks a lot, it did work on development branch!  (pip install git+https://github.com/tapegoji/PySpice@development)

I got some warning at this line:  `spice_source = SpiceSource(source=source, title_line=False) `

```
WARNING: Token 'STRING' defined, but not used
WARNING: There is 1 unused token
Generating LALR tables
WARNING: 57 shift/reduce conflicts
WARNING: 1 reduce/reduce conflict
WARNING: reduce/reduce conflict in state 58 resolved using rule (expression_list_space -> expression)
WARNING: rejected rule (expression_list_comma -> expression) in state 58
``` 
I do not know what they mean / if they are relevant

0range0range, 2025-04-17:
Unfortunately when parsing circuit of interest another error occured: 

```
2025-04-17 13:04:03,200 - PySpice.Spice.Parser.Parser.SpiceParser.p_error - ERROR - Syntax error at ','

---------------------------------------------------------------------------
NameError                                 Traceback (most recent call last)
File ......venvs/ngspice-python-export/lib/python3.13/site-packages/PySpice/Spice/Parser/HighLevelParser.py:1084, in SpiceSource.parse(self)
   1083 try:
-> 1084     ast = self._parser.parse(line.command)
   1085     self._ast_lines.append(ast)

File ......Python/venvs/ngspice-python-export/lib/python3.13/site-packages/PySpice/Spice/Parser/Parser.py:587, in SpiceParser.parse(self, line)
    586 # self._parser.defaulted_states = {}
--> 587 return self._parser.parse(line, lexer=self._lexer, debug=False)

File ......Python/venvs/ngspice-python-export/lib/python3.13/site-packages/ply/yacc.py:333, in LRParser.parse(self, input, lexer, debug, tracking, tokenfunc)
    332 else:
--> 333     return self.parseopt_notrack(input, lexer, debug, tracking, tokenfunc)

File ......Python/venvs/ngspice-python-export/lib/python3.13/site-packages/ply/yacc.py:1201, in LRParser.parseopt_notrack(self, input, lexer, debug, tracking, tokenfunc)
   1200 self.state = state
-> 1201 tok = call_errorfunc(self.errorfunc, errtoken, self)
   1202 if self.errorok:
   1203     # User must have done some kind of panic
   1204     # mode recovery on their own.  The
   1205     # returned token is the next lookahead

File ......Python/venvs/ngspice-python-export/lib/python3.13/site-packages/ply/yacc.py:192, in call_errorfunc(errorfunc, token, parser)
    191 _restart = parser.restart
--> 192 r = errorfunc(token)
    193 try:

File ......Python/venvs/ngspice-python-export/lib/python3.13/site-packages/PySpice/Spice/Parser/Parser.py:499, in SpiceParser.p_error(self, p)
    498     self._logger.error(f"Syntax error at '{p.value}'")
--> 499     raise NameError('Syntax Error')
    500 else:

NameError: Syntax Error

During handling of the above exception, another exception occurred:

ParseError                                Traceback (most recent call last)
Cell In[26], line 1
----> 1 parser1 = SpiceSource(source=source_divider, title_line=False)

File  .....Python/venvs/ngspice-python-export/lib/python3.13/site-packages/PySpice/Spice/Parser/HighLevelParser.py:983, in SpiceSource.__init__(self, source, title_line)
    981 self.reset()
    982 if source is not None:
--> 983     self.parse_string(source, title_line)

File ......Python/venvs/ngspice-python-export/lib/python3.13/site-packages/PySpice/Spice/Parser/HighLevelParser.py:1180, in SpiceSource.parse_string(self, source, title_line)
   1178 generator = enumerate(source.splitlines())
   1179 self.read(generator, title_line)
-> 1180 self._parse()

File ......Python/venvs/ngspice-python-export/lib/python3.13/site-packages/PySpice/Spice/Parser/HighLevelParser.py:1172, in SpiceSource._parse(self)
   1171 def _parse(self) -> None:
-> 1172     self.parse()
   1173     self.analyse()

File ......Python/venvs/ngspice-python-export/lib/python3.13/site-packages/PySpice/Spice/Parser/HighLevelParser.py:1088, in SpiceSource.parse(self)
   1086     self._logger.debug(os.linesep + str(ast.pretty_print()))
   1087 except NameError as e:
-> 1088     raise ParseError(str(e))

ParseError: Syntax Error
```

Here is the source of the circuit: 

* Qucs 25.1.1  /home/serhii/3_Resources/QUCS/projects/Tranistor-tester_prj/Full-tester.sch
.INCLUDE "/usr/share/qucs-s/xspice_cmlib/include/ngspice_mathfunc.inc"

.SUBCKT SpiceOpamp_LM358 	gnd 1 2 3 4 5
*
C1   11 12 5.544E-12
C2    6  7 20.00E-12
DC    5 53 DX
DE   54  5 DX
DLP  90 91 DX
DLN  92 90 DX
DP    4  3 DX
EGND 99  0 POLY(2) (3,0) (4,0) 0 .5 .5
FB    7 99 POLY(5) VB VC VE VLP VLN 0 15.91E6 -20E6 20E6 20E6 -20E6
GA    6  0 11 12 125.7E-6
GCM   0  6 10 99 7.067E-9
IEE   3 10 DC 10.04E-6
HLIM 90  0 VLIM 1K
Q1   11  2 13 QX
Q2   12  1 14 QX
R2    6  9 100.0E3
RC1   4 11 7.957E3
RC2   4 12 7.957E3
RE1  13 10 2.773E3
RE2  14 10 2.773E3
REE  10 99 19.92E6
RO1   8  5 50
RO2   7 99 50
RP    3  4 30.31E3
VB    9  0 DC 0
VC 3 53 DC 2.100
VE   54  4 DC .6
VLIM  7  8 DC 0
VLP  91  0 DC 40
VLN   0 92 DC 40
.MODEL DX D(IS=800.0E-18)
.MODEL QX PNP(IS=800.0E-18 BF=250)
.ENDS
  
.SUBCKT Power_amp_arduino_subscheme Vout gnd Vin VCC 
QX2N2222A_1 VCC Vopamp Vout QMOD_X2N2222A_1 AREA=1
.MODEL QMOD_X2N2222A_1 npn (Is=14.34F Nf=1 Nr=1 Ikf=0.2847 Ikr=0 Vaf=74.03 Var=0 Ise=14.34F Ne=1.307 Isc=0 Nc=2 Bf=255.9 Br=6.092 Rbm=0 Irb=0 Rc=1 Re=0 Rb=10 Cje=22.01P Vje=0.75 Mje=0.377 Cjc=7.306P Vjc=0.75 Mjc=0.3416 Xcjc=1 Cjs=0 Vjs=0.75 Mjs=0 Fc=0.5 Tf=411.1P Xtf=3 Vtf=0 Itf=0.6 Tr=46.91N Kf=0 Af=1 Ptf=0 Xtb=1.5 Xti=3 Eg=1.11 Tnom=26.85 )
RG 0 _net2  100K
XOP1 0  Vin _net2 VCC 0 Vopamp SpiceOpamp_LM358
Rf _net2 Vout  1K 
RG1 Vin 0  100K
.ENDS
Rsh Vin _net0  1K
V2 VCC 0 DC 8
V4 Vin 0 DC 2
XSUB1 _net4 0 _net0 VCC Power_amp_arduino_subscheme
QX2N2222A_1 _net5 Vbase 0 QMOD_X2N2222A_1 AREA=1
.MODEL QMOD_X2N2222A_1 npn (Is=14.34F Nf=1 Nr=1 Ikf=0.2847 Ikr=0 Vaf=74.03 Var=0 Ise=14.34F Ne=1.307 Isc=0 Nc=2 Bf=255.9 Br=6.092 Rbm=0 Irb=0 Rc=1 Re=0 Rb=10 Cje=22.01P Vje=0.75 Mje=0.377 Cjc=7.306P Vjc=0.75 Mjc=0.3416 Xcjc=1 Cjs=0 Vjs=0.75 Mjs=0 Fc=0.5 Tf=411.1P Xtf=3 Vtf=0 Itf=0.6 Tr=46.91N Kf=0 Af=1 Ptf=0 Xtb=1.5 Xti=3 Eg=1.11 Tnom=26.85 )
VIb _net4 OutV DC 0
Rsh2 Vbase OutV  1K

.endc
.END
```

is it related to this line: `EGND 99  0 POLY(2) (3,0) (4,0) 0 .5 .5`?

Kreijstal, 2025-04-17:
exactly, the parser has lots of errors, therefore it is undesirable to convert the circuit to spice then parse it, we dont want to parse it we want to simulate it, and parser just means throwing errors... therefore the previous method was better

tapegoji, 2025-04-17:
> [@tapegoji](https://github.com/tapegoji) if I understand this corerectly. you conver this to spice, then convert it back to circuit, this is undesirable, I'd like to use the pyspice api as it was, without going through this hacks. Otherwise I'd just use spice without pyspice...

there are applications, when you bring in your raw_spice and convert it to circuit objects. then you can modify and simulate.

tapegoji, 2025-04-17:
your E voltage source, does not seem to be in ngspice format. what format is this?

0range0range, 2025-04-17:
@tapegoji 
> your E voltage source, does not seem to be in ngspice format. what format is this?

If you mean this part `EGND 99  0 POLY(2) (3,0) (4,0) 0 .5 .5`, the whole LM358 subcircuit is a part of the standart component library provided by Qucs-S as installed. I drew the schematic in Qucs-S and then exported netlist. 

@Kreijstal 
> exactly, the parser has lots of errors, therefore it is undesirable to convert the circuit to spice then parse it, we dont want to parse it we want to simulate it, and parser just means throwing errors... therefore the previous method was better

By this, you mean it is better to construct the circuit from within PySpice? If I understand correctly, I still need to pass the LM358 model as raw_spice, facing the same error.

tapegoji, 2025-04-17:
@0range0range 

Got it, it's a non-linear voltage source. There are two issues, that I have found. Parser has some issues. and second the high-lever parser does not know that it's a non-linear source. Seems to be a challenge but can be fixed.

Kreijstal, 2025-04-18:
> [@tapegoji](https://github.com/tapegoji)
> 
> > your E voltage source, does not seem to be in ngspice format. what format is this?
> 
> If you mean this part `EGND 99 0 POLY(2) (3,0) (4,0) 0 .5 .5`, the whole LM358 subcircuit is a part of the standart component library provided by Qucs-S as installed. I drew the schematic in Qucs-S and then exported netlist. I have no idea :)
> 
> [@Kreijstal](https://github.com/Kreijstal)
> 
> > exactly, the parser has lots of errors, therefore it is undesirable to convert the circuit to spice then parse it, we dont want to parse it we want to simulate it, and parser just means throwing errors... therefore the previous method was better
> 
> By this, you mean it is better to construct the circuit from within PySpice? If I understand correctly, I still need to pass the LM358 model as raw_spice, facing the same error.

That's tapegoji workaround... He wants to construct the ngspice then parse it, I think that's is inefficient

tapegoji, 2025-04-18:
@0range0range I think It's okay now. Look at fix_non_linear_E branch and try to test it. Let me know

0range0range, 2025-04-23:
@tapegoji  

Hi, sorry for delayed responce. 

Thanks! Parsing is working, I am able to print the circuit as it should be. 

When I try to move to simulation I still get an error: 
```
source_divider = open(str(Path("../test/transistor-tester-netlist.cir"))).read()
parser = SpiceSource(source=source_divider, title_line=False)
test_circuit = Translator.Builder().translate(parser)
simulator = test_circuit.simulator(temperature=25)

---------------------------------------------------------------------------
NameError                                 Traceback (most recent call last)
Cell In[18], line 1
----> 1 simulator = test_circuit.simulator(temperature=25)

File ~/3_Resources/Programming/Python/venvs/ngspice-python-export/lib/python3.13/site-packages/PySpice/Spice/Netlist.py:745, in Circuit.simulator(self, *args, **kwargs)
    743 def simulator(self, *args, **kwargs):
    744     # return CircuitSimulator.factory(self, *args, **kwargs)
--> 745     raise NameError("Deprecated API")

NameError: Deprecated API
```

With Pyspice 1.5 this approach worked. What is the way to start simulation with PySpice 1.6? 

All in all, I think this can be a workaround for original issue. 

@Kreijstal 
Just to understand the efficiency issue. You mean that I should recreate this, as a Subcircuit class, instead of trying to parse from outside? It might get quite complcated with other SPICE models. Some are even bigger. I think it would be probably even better to parse such models separately in SubCircuit class, without manually typing out the elements.

```
.SUBCKT SpiceOpamp_LM358 	gnd 1 2 3 4 5
*
C1   11 12 5.544E-12
C2    6  7 20.00E-12
DC    5 53 DX
DE   54  5 DX
DLP  90 91 DX
DLN  92 90 DX
DP    4  3 DX
EGND 99  0 POLY(2) (3,0) (4,0) 0 .5 .5
FB    7 99 POLY(5) VB VC VE VLP VLN 0 15.91E6 -20E6 20E6 20E6 -20E6
GA    6  0 11 12 125.7E-6
GCM   0  6 10 99 7.067E-9
IEE   3 10 DC 10.04E-6
HLIM 90  0 VLIM 1K
Q1   11  2 13 QX
Q2   12  1 14 QX
R2    6  9 100.0E3
RC1   4 11 7.957E3
RC2   4 12 7.957E3
RE1  13 10 2.773E3
RE2  14 10 2.773E3
REE  10 99 19.92E6
RO1   8  5 50
RO2   7 99 50
RP    3  4 30.31E3
VB    9  0 DC 0
VC 3 53 DC 2.100
VE   54  4 DC .6
VLIM  7  8 DC 0
VLP  91  0 DC 40
VLN   0 92 DC 40
.MODEL DX D(IS=800.0E-18)
.MODEL QX PNP(IS=800.0E-18 BF=250)
.ENDS
```

Kreijstal, 2025-04-23:
@0range0range no what I meant is if you have already defined the circuit with pyspice, pyspice shouldn't reparse it. of course, if you haven't done it in pyspice, there is no need to "recreate it". But my comment was meant to @tapegoji  mostly

tapegoji, 2025-04-23:
> [@tapegoji](https://github.com/tapegoji)
>
> Hi, sorry for delayed responce.
>
> Thanks! Parsing is working, I am able to print the circuit as it should be.
>
> When I try to move to simulation I still get [...]

@0range0range In 1.6 there is a simulation factory class that needs to be called. 

Here is an example. 
from PySpice import Circuit, Simulator
simulator = Simulator.factory()
simulation = simulator.simulation(circuit, temperature=25, nominal_temperature=25)

You can look at the examples for more info.

tapegoji, 2025-10-29:
Pyspice 1.6 has a parser which is vastly differnet than the old one. it brings a lot of new features, but its in a very early stage. I have made a repo and fixed a bunch. I made PR and no response. Therfore, I made a new package called InSpice. check out my repo and use that. a lot fo these issues are solved there, I hope