PySpice-org/PySpice #294: How can we measure voltage between two points to write expression in behavioral voltage source?

prajapatisantlal, 2021-09-10:
I writing an expression in a subcircuit with pyspice as V(x,y) where x and y are two nodes. How to fix it?

benedictjones, 2021-09-13:
It is quite hard to understand exactly what you are asking? Can you give an example?

The first thing that came to mind for me is the expressions you might write for a B source.
For example, a step function:
      
Consider two nodes "in" and "out". Lets write out current defined equation (in this case an **if statement**) based on the voltages at these nodes:
`i_express = "((v(in)-v(out))>0) ? 1 : 0"` # _i.e., if V(in,out) > 0 then i=1, else i=0_
 Now, we use this to define the current of the B source component:
`self.B('myB1', 'in', 'out', i=i_express)`

As far as I remember, when writing in the raw spice (which is what you are basically doing here) the voltage between two nodes is **not** defined as `V(x,y) `but as `v(x)-v(y)`. Though you can look at the NGSpice manual to check this.
I hope this helped?

santlal123, 2021-09-19:
Thank you benedictjones. I exactally want this.