PySpice-org/PySpice #300: NgSpice shared instance cannot quit

zhaofeng-shu33, 2021-11-04:
### Environment (OS, Python version, PySpice version, simulator)
Windows 10, Python 38, v1.5, ngspice
### Expected Behaviour
The code should not raise exception.
### Actual Behaviour
```
Traceback (most recent call last):
  File "test_ngspice_interpreter.py", line 73, in <module>
    ngspice.quit()
  File "C:\Users\z50019165\AppData\Local\Programs\Python\Python38\lib\site-packages\PySpice\Spice\NgSpice\Shared.py", line 1006, in quit
    return self.exec_command('quit')
  File "C:\Users\z50019165\AppData\Local\Programs\Python\Python38\lib\site-packages\PySpice\Spice\NgSpice\Shared.py", line 848, in exec_command
    raise NameError("ngSpice_Command '{}' returned {}".format(command, rc))
NameError: ngSpice_Command 'quit' returned 1
```
### Steps to reproduce the behaviour
Run [this example](https://pyspice.fabrice-salvaire.fr/releases/v1.5/examples/ngspice-shared/ngspice-interpreter.html)
with uncommented last line `ngspice.quit()`.

zhaofeng-shu33, 2021-11-08:
Same problem persists in Ubuntu 18.04.

thejackal360, 2022-01-19:
This appears to be an ngspice issue (https://sourceforge.net/p/ngspice/discussion/133842/thread/8b0d3c1d/?limit=25):

"ngSpice_Command returns 1 if the internal function shared_exit() has been called to terminate ngspice.
This may happen after having sent the 'quit' command or due to an error during execution in ngspice.dll."

I would try out the following patch:

https://sourceforge.net/u/thejackal360/ngspice/ci/28839e88384971e36c1374c7a3738d560e4f95b0/