PySpice-org/PySpice #319: Scattering Parameters

GleidsonLeite, 2022-04-23:
How can i extract S-parameters using PySpice? Is there any example to do that?

matthuszagh, 2022-06-12:
It appears that [s-parameter analysis is coming to ngspice](https://sourceforge.net/p/ngspice/discussion/127605/thread/7a2655c86f/). Once that's in stable, I expect it wouldn't be too much work to get this into pyspice.

If you don't want to wait for that you can always create something yourself. You'll need some sort of signal separation device. I'd probably look at the [Wenzel low-frequency circulator](http://www.qrp.gr/circulator/RFDesign3.pdf) and use ideal opamps so you're not bandwidth-limited. Worth also noting that a lot of people use other tools than spice for RF simulation (e.g., QUCS) or a proper EM simulator (openems and meep are open-source options, but they carry pretty steep learning curves).

matthuszagh, 2022-06-16:
[Now released](https://sourceforge.net/projects/ngspice/files/ng-spice-rework/37/). Maybe we turn this into an issue to add support for SP simulation?