PySpice-org/PySpice #325: Unable to use Voltage Controlled Voltage Source (VCVS)

nataraj-pinakapani, 2022-06-04:
Environment (Ubuntu 20.04, Python 3.8.10, PySpice version, Simulator --> NgSpice)

Issue: Unable to use Voltage Controlled Voltage Source (VCVS)

Description: If I use the VCVS, I get AttributeError. 

PySpice Code: 
--------------------
libraries_path ='/home/nataraj/projects/designmyic/cad/sky130_pdk_pyspice/open_pdks/sky130/sky130A/libs.tech/ngspice/'
spice_library = SpiceLibrary(libraries_path)
ngspice = NgSpiceShared.new_instance()

circuit =  Circuit('test_circuit')
circuit.include(spice_library['sky130_all_tt'])

circuit.X('M1', 'sky130_fd_pr__nfet_01v8', 'D', 'G', 'S' ,'B', w=1, l=1, nf=1, m=1)
circuit.V('SS', 'S', circuit.gnd, 0)
circuit.V('GS', 'G', 'S', 1)
circuit.E('1', 'G', 0, 'D', 0, 1)

Error:
-------
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/PySpice/Spice/Netlist.py", line 918, in __getattr__
    return self.__getitem__(attribute_name)
  File "/usr/local/lib/python3.8/dist-packages/PySpice/Spice/Netlist.py", line 912, in __getitem__
    raise IndexError(attribute_name)   # KeyError
IndexError: E

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/nataraj/projects/designmyic/tee-internal/codes/app_codes/pyspice_codes/test.py", line 29, in <module>
    circuit.E('1', 'G', 0, 'D', 0, 1)
  File "/usr/local/lib/python3.8/dist-packages/PySpice/Spice/Netlist.py", line 920, in __getattr__
    raise AttributeError(attribute_name)
AttributeError: E

pkneuper1, 2022-06-22:
It looks like it is not implemented yet. You can use raw_spice instead.

User-0007, 2022-10-19:
You can declare it by using  circuit.VCVS('1', 'G', 0, 'D', 0, 1) instead

nataraj-pinakapani, 2022-10-23:
> circuit.VCVS('1', 'G', 0, 'D', 0, 1)

Thank you very much

closed 2022-10-23 by nataraj-pinakapani