PySpice-org/PySpice #326: Value after * must be an iterable, not NoneType

tbarton1197, 2022-06-13:
### Environment (OS, Python version, PySpice version, simulator)
Spyder, Windows 11, pyspice v1.5, ngspice

I'm running code nearly copied from the example here but with different components 

https://pyspice.fabrice-salvaire.fr/releases/v1.3/examples/advanced-usages/netlist-manipulations.html

`class neuron(SubCircuitFactory):
    __name__ = 'neuron'
    __nodes__ = ('in', 'out')
    def __init__(self):
        super().__init__(self.__name__, self.__nodes__, [])
        
        self.include(spice_library['1N4148'])
        self.X('D1','1N4148', 'in', self.gnd)
        self.X('D2','1N4148', self.gnd, 'in')
        self.VCVS(1, 'out', self.gnd, 'in', self.gnd, voltage_gain=4)`

But when I instantiate the class, I get this error:TypeError: Value after * must be an iterable, not NoneType

`from PySpice.Spice.Netlist import Circuit, SubCircuit, SubCircuitFactory
from PySpice.Unit import *
import PySpice.Logging.Logging as Logging
import subcircuits as tbsc

logger = Logging.setup_logging()
circuit = Circuit('one_neuron')
ground = circuit.gnd

circuit.subcircuit(neuron())`

here is the full traceback

Traceback (most recent call last):

  File "C:\Users\tbart\Miniconda3\lib\site-packages\spyder_kernels\py3compat.py", line 356, in compat_exec
    exec(code, globals, locals)

  File "c:\users\tbart\documents\grad school\research\eqprop\v_pyspice\setup.py", line 21, in <module>
    circuit.subcircuit(tbsc.BasicOperationalAmplifier())

  File "C:\Users\tbart\Documents\Grad School\Research\eqprop\v_pyspice\subcircuits.py", line 44, in __init__
    super().__init__()

  File "C:\Users\tbart\Miniconda3\lib\site-packages\PySpice\Spice\Netlist.py", line 1125, in __init__
    super().__init__(self.NAME, *self.NODES, **kwargs)

TypeError: Value after * must be an iterable, not NoneType

pkneuper1, 2022-06-21:
This error was resolved for me after I removed SubCircuit from my import list (only import SubCircuitFactory)

kdre, 2023-10-21:
Have the same issue when trying to execute with PySpice 1.5: 
https://github.com/OlzhasT/PySpiceCircuits-/blob/master/adc_dac_ideal_subcircuit_class_test.py

Ans solutions?

AshuPat, 2023-11-03:
Possible solution is as follows. 
Source: [EngineeringThings] [https://youtu.be/znaERIx1tL8]

```
from PySpice.Spice.Netlist import SubCircuit
from PySpice.Spice.Netlist import Circuit
from PySpice.Unit import *

class MySubCir(SubCircuit):
    __nodes__=('t_in','t_out')
    def __init__(self, name, r=1@u_kOhm):
        SubCircuit.__init__(self,name,*self.__nodes__)
        self.R(1,'t_in','t_out',r)
        return
circuit = Circuit('Main Circuit')
circuit.subcircuit(MySubCir('sub1',r=1@u_kOhm))
circuit.X(1,'sub1', 1 ,circuit.gnd)
```