PySpice-org/PySpice #339: Command 'run' failed following Internal Device Parameters documentation

KevinLiebergen, 2023-02-24:
### Environment (OS, Python version, PySpice version, simulator)
Ubuntu 20.04.5 LTS, Python 3.10,  PySpice 1.5

### Actual Behaviour

The last commented line from
https://pyspice.fabrice-salvaire.fr/releases/v1.5/examples/advanced-usages/internal-device-parameters.html
(compute the DC transfer function) failed

```
NgSpiceCommandError                       Traceback (most recent call last)
Cell In[20], line 2
      1 simulator = circuit.simulator(temperature=25, nominal_temperature=25)
----> 2 analysis = simulator.dc(Vdd=slice(0, 5, .1)) # Fixme: ,Vsig=slice(1, 5, 1)

File ~/anaconda3/envs/tfg_1.4/lib/python3.9/site-packages/PySpice/Spice/Simulation.py:1151, in CircuitSimulator.dc(self, *args, **kwargs)
   1150 def dc(self, *args, **kwargs):
-> 1151     return self._run('dc', *args, **kwargs)

File ~/anaconda3/envs/tfg_1.4/lib/python3.9/site-packages/PySpice/Spice/NgSpice/Simulation.py:118, in NgSpiceSharedCircuitSimulator._run(self, analysis_method, *args, **kwargs)
    116 # Fixme: Error: circuit not parsed.
    117 self._ngspice_shared.load_circuit(str(self))
--> 118 self._ngspice_shared.run()
    119 self._logger.debug(str(self._ngspice_shared.plot_names))
    120 self.reset_analysis()

File ~/anaconda3/envs/tfg_1.4/lib/python3.9/site-packages/PySpice/Spice/NgSpice/Shared.py:1168, in NgSpiceShared.run(self, background)
   1165 #  in the background thread and wait until the simulation is done
   1167 command = 'bg_run' if background else 'run'
-> 1168 self.exec_command(command)
   1170 if background:
   1171     self._is_running = True

File ~/anaconda3/envs/tfg_1.4/lib/python3.9/site-packages/PySpice/Spice/NgSpice/Shared.py:842, in NgSpiceShared.exec_command(self, command, join_lines)
    839     raise NameError("ngSpice_Command '{}' returned {}".format(command, rc))
    841 if self._error_in_stdout or self._error_in_stderr:
--> 842     raise NgSpiceCommandError("Command '{}' failed".format(command))
    844 if join_lines:
    845     return self.stdout

NgSpiceCommandError: Command 'run' failed
```

Why the last stable version doesn't work? It seems the circuit is not parsed, how do I parse it?
It also happens with PySpice 1.4, although the same line is uncommented.

khodjawalid, 2025-01-24:
Same problem here after two years, i don't know if you got you answer or not ?

khodjawalid, 2025-01-24:
> pyspice doesnt parse the circuit, ngspice does

Thanks for your answer but what is the solution to simulate the circuit?

Kreijstal, 2025-01-24:
> Thanks for your answer but what is the solution to simulate the circuit?
> 
> Le ven. 24 janv. 2025 à 13:38, Kreijstal ***@***.***> a
> écrit :
> […](#)

try the code first in ngspice, no?

khodjawalid, 2025-01-24:
I work on pyspice using python for my application