PySpice-org/PySpice #340: Pyspice for commercials simulators

leochand101, 2023-03-18:
I wanted to use the pyspice framework to run spectre, I see some people have ported it to standard technologies. "
Anybody have a branch where they have made updates related to spectre/Hspice enabled pyspice ?

Or if someone has ideas of which files need modifications and testing, I can give it a try and commit it here.

leochand101, 2023-06-14:
@infinitymdm Marcus I am currently using the Pyspectre for my requirements, which is a reduced but sufficient version only for spectre and reads in nutbin format output into pandas. Not sure if that is a simpler interface to add Hspice, if there is an interactive version to run design space optimization in loops though it could probably work.

infinitymdm, 2023-06-14:
Just based on how simple the [Xyce interface](https://github.com/PySpice-org/PySpice/tree/master/PySpice/Spice/Xyce) is, I'd suspect setting something up for hspice wouldn't be too terribly difficult. I'm not aware of a fork that has this functionality at present, but it may be something I end up implementing if I have the time.