PySpice-org/PySpice #350: AWS Lambda: OSError: cannot load library 'libngspice.so'

GiovanniLeo, 2023-07-31:
**OS: Amazon Linux 2
Python Version: 3.10
PySpice Version: 1.5**

I'm attempting to use the PySpice library within an AWS Lambda function, but encountered some issues along the way. I diligently followed the installation guide for Linux using PyPl, as it was the only viable option within this environment. However, when I attempt to run the command` pyspice-post-installation --check-install` , it presents the following error:

`"OSError: cannot load library 'libngspice.so': libngspice.so: cannot open shared object file: No such file or directory. Additionally, ctypes.util.find_library() did not manage to locate a library called 'libngspice.so.'"`

Can someone guide me on how to resolve this issue? I greatly appreciate your help!

cyber-g, 2023-08-09:
Did you try to install libngspice ? 

If I understand, Amazon linux user dnf as package manager. Please read the section for RPM distributions : 
https://pyspice.fabrice-salvaire.fr/releases/v1.6/installation.html#on-linux

GiovanniLeo, 2023-08-10:
In Amazon Linux 2, the package manager is **yum**. I find a way to install it on Amazon Linux 2 by containerizing the application to create a Lambda Container, if you want I can share my solution.

cyber-g, 2023-08-11:
Yes I am interested, mostly for sale of curiosity, because AWS is totally new for me. 
I would like to see how much it differs from a previous docker implementation for (older) debian distribution I did : 
https://github.com/cyber-g/debian-pyspice-semi-custom
https://github.com/cyber-g/debian-pyspice-std

GiovanniLeo, 2023-08-15:
```
FROM public.ecr.aws/lambda/python:3.10.2023.08.02.10
LABEL authors="giovannileo"

# Installing all the developer tools
RUN yum -y update
RUN yum -y groupinstall "Development Tools"
RUN yum -y install openssl11 openssl11-devel bzip2-devel libffi-devel
RUN yum -y install wget
#Intsalling required sofware for ngspice
RUN yum -y install libc libcx libgcc1 libssp libstdc++6 libstdc++ libsupc++6 libsupc++ libgcc-fwd
RUN yum -y install  gcc make rpm-build libtool hwloc-devel libX11-devel libXt-devel libedit-devel libical-devel ncurses-devel perl postgresql-devel python-devel tcl-devel tk-devel swig expat-devel openssl-devel libXext libXft --skip-broken

#downloading ngspice
RUN wget -O ngspice-34.tar.gz https://sourceforge.net/projects/ngspice/files/ng-spice-rework/old-releases/34/ngspice-34.tar.gz/download
RUN tar xvf ngspice-34.tar.gz
#intsalling ngspice -> look at source
RUN cd ngspice-34 &&  ./configure --with-ngshared --with-tcl=tcldir --with-editline=no --with-readline=no --prefix=/var/task && make && make install
RUN cd ..
RUN rm -rf ngspice-34.tar.gz

#Loading lambda
ADD lambda ./
#Installing requirements
RUN pip install -r requirements.txt
RUN pip install opencv-python-headless

# Set the CMD to your handler (could also be done as a parameter override outside of the Dockerfile)
CMD [ "index.handler" ]
```

In this setup, I'm utilizing PySpice 1.5 alongside various dependencies specified in the requirements.txt file. As previously mentioned, I'm operating within an environment running Amazon Linux 2, which employs the yum package manager. In this specific context, I'm focused on installing NgSpice version 34, as it's the mandated version for our purposes.

The installation process involves configuring NgSpice 34 to be used as shared libraries. The flags I've provided in the following command are crucial for this setup: `/configure --with-ngshared --with-tcl=tcldir --with-editline=no --with-readline=no --prefix=/var/task`. Each flag serves a distinct purpose, such as enabling the use of shared libraries for the OS and setting a specific installation directory using the prefix.

The necessary instructions for this installation can be found within the "docs," specifically the INSTALL file available on SourceForge, as well as through various discussions on relevant forums.

Within the Dockerfile, the sections of utmost significance are indicated by the comments `# Installing all the developer tools` and `# Downloading ngspice`. These sections are pivotal since the subsequent steps pertain directly to the lambda function.

In the first section, I'm ensuring all prerequisites required for NgSpice are installed. The subsequent section involves the download and installation of the NgSpice tool itself. This comprehensive approach ensures the successful integration of NgSpice within the lambda function.