PySpice-org/PySpice #352: Using `circuit.simulator(simulator='ngspice-subprocess',...)` breaks PySpice execution

cyber-g, 2023-09-08:
### Environment (OS, Python version, PySpice version, simulator)
Ubuntu 20.04.6 LTS
Python 3.8.10
PySpice version: '1.5'
Simulator: NGSpice

### Expected Behaviour
No error message ; elements values should be printed

### Actual Behaviour
PySpice stops with the following output: 

```bash
me@compute:PySpice/examples/resistor$ python3 voltage-divider.py 
2023-09-08 19:08:14,060 - PySpice.Spice.Netlist.Node.__init__ - WARNING - Node name 'in' is a Python keyword
2023-09-08 19:08:14,071 - PySpice.Spice.NgSpice.Server.SpiceServer.__call__ - INFO - Start the spice subprocess
Traceback (most recent call last):
  File "voltage-divider.py", line 27, in <module>
    analysis = simulator.operating_point()
  File "git-dir/PySpice/PySpice/Spice/Simulation.py", line 1194, in operating_point
    return self._run('operating_point', *args, **kwargs)
  File "git-dir/PySpice/PySpice/Spice/NgSpice/Simulation.py", line 76, in _run
    raw_file = self._spice_server(spice_input=str(self))
  File "git-dir/PySpice/PySpice/Spice/NgSpice/Server.py", line 162, in __call__
    return RawFile(stdout, number_of_points)
  File "git-dir/PySpice/PySpice/Spice/NgSpice/RawFile.py", line 170, in __init__
    raw_data = self._read_header(stdout)
  File "git-dir/PySpice/PySpice/Spice/NgSpice/RawFile.py", line 192, in _read_header
    self.circuit_name = self._read_header_field_line(header_line_iterator, 'Circuit')
  File "git-dir/PySpice/PySpice/Spice/RawFile.py", line 243, in _read_header_field_line
    raise NameError("Expected label %s instead of %s" % (expected_label, label))
NameError: Expected label Circuit instead of No compatibility mode selected
me@compute:PySpice/examples/resistor$ 
```

The failure happens for any kind of circuit but is demonstrated below with one of the builtin example. 

### Steps to reproduce the behaviour
Open example `examples/resistor/voltage-divider.py` and replace line [`L25`](https://github.com/PySpice-org/PySpice/blob/master/examples/resistor/voltage-divider.py#L25):
```python
simulator = circuit.simulator(temperature=25, nominal_temperature=25)
```
with
```python
simulator = circuit.simulator(simulator='ngspice-subprocess',temperature=25, nominal_temperature=25)
```
Then run : 
```bash
python3 voltage-divider.py
```

cyber-g, 2023-09-08:
I have investigated the problem. It comes from an evolution on NGSpice side that begun on 2021-01-01 : 
https://sourceforge.net/p/ngspice/ngspice/ci/1234c3bdf8a48e6201a3ecf03e10396657fa96e8/tree//src/frontend/inpcom.c?diff=164d3dd20c6008d842bd4ae34d63d3ae22923e68

From that moment, `inpcom.c` began to display a message at the very beginning of the output of the simulation mode. 
The message has been latter moved to [`inpcompat.c`](https://sourceforge.net/p/ngspice/ngspice/ci/master/tree/src/frontend/inpcompat.c). 

The message can be easily shown with the following command from the terminal : 
```bash
echo "bla" | ngspice -s

No compatibility mode selected!

Circuit: bla

Note: No ".plot", ".print", or ".fourier" lines; no simulations run
```
which mimics what is done in [Server.py](https://github.com/PySpice-org/PySpice/blob/master/PySpice/Spice/NgSpice/Server.py#L146)

If one sets an `spinit` file containing a compatibility setting, for example: 
```bash
set ngbehavior=hsa
```
and configures the `SPICE_SCRIPTS` environment variable to use this `spinit` file then you get: 
```bash
echo "bla" | ngspice -s

Compatibility modes selected: hs a

Circuit: bla

Doing analysis at TEMP = 27.000000 and TNOM = 27.000000
```

Please note that the message is not an error; it is just a *note* given by NGSpice. 

To fix the issue, one may want to update : 
[`_read_header`](https://github.com/PySpice-org/PySpice/blob/master/PySpice/Spice/NgSpice/RawFile.py#L178)

to contain something like:
```python
self.note = self._read_header_field_line(header_line_iterator, 'Note')
```
(only valid for the latest versions of NGSpice ([later than 2022-03-14](https://sourceforge.net/p/ngspice/ngspice/ci/4cc6222230e05dee45845a0df7e7b92275947279/tree//src/frontend/inpcom.c?diff=696736b2b3ee915379979b133ea76a7f741337f6))

I need to check if this first idea for fix is valid.

cyber-g, 2023-09-08:
Please note, that there is no issue when using the `circuit.simulator(simulator='ngspice-shared')` option instead

KolosKoblasz, 2024-05-31:
Hi @cyber-g !

I'm trying to run ngspice sims with PySpice on Ubuntu, but I'm stuck on an issue.
Could you help me to fix it?

I created the environment variable in my .bashrc:
SPICE_SCRIPTS="/home/kolos/spinit"
echo $SPICE_SCRIPTS /home/kolos/spinit

but ngspice can't find the spinit file:

Note: can't find the initialization file spinit.

```
Note: No compatibility mode selected!

Circuit: bla

Doing analysis at TEMP = 27.000000 and TNOM = 27.000000

Using SPARSE 1.3 as Direct Linear Solver
```

cyber-g, 2024-11-20:
Dear @KolosKoblasz 

Your issue is a little bit off topic of this track issue. However, it should be easy to fix with following explanation. 
The environment variable `SPICE_SCRIPTS` should contain the path of the directory that contains the `spinit` file. 

This is documented here : [12.5 Standard configuration file spinit
](https://ngspice.sourceforge.io/docs/ngspice-html-manual/manual.xhtml#magicparlabel-23856)

> The path may be overridden by setting the environmental variable SPICE_SCRIPTS to a path where spinit is located.

So if you have your `spinit` file in `/home/kolos/`, then you have to put in your `.bashrc`: 

```bash
export SPICE_SCRIPTS="/home/kolos/"
```

davidcastells, 2025-12-14:
I am not sure if this is fixed or not. Sorry, I am a little bit lost with checking the commits 
In my system I fixed by adding a line

    # dummy code to avoid the crash
        self._read_header_field_line(header_line_iterator, 'Note')

before 
        self.circuit_name = self._read_header_field_line(header_line_iterator, 'Circuit')

in PySpice/Spice/NgSpice/RawFile.py