PySpice-org/PySpice #357: Bode diagram plot issue

mbriday, 2023-10-15:
Hi, 
Thanks for this great python module!!

### Environment (OS, Python version, PySpice version, simulator)

Linux Debian 12.1 with Python 3.11.2, pyspice v1.5, installed from pip3 (using a venv)

### Actual Behaviour

There is an issue with matplotlib (installed using pip3) , the example `filter/low-pass-rc-filter.py` does not work. It complains about 'basex' and stops the script:

```python
Traceback (most recent call last):
  File "spice/examples/filter/low-pass-rc-filter.py", line 44, in <module>
    bode_diagram(axes=axes,
[...]
TypeError: LogScale.__init__() got an unexpected keyword argument 'basex'
```

### Steps to reproduce the behaviour

just run the `filter/low-pass-rc-filter.py` example, using Python 3.11.2 and  pip3 (matplotlib, pyspice).

### workaround

from [semilog matplotlib doc](https://matplotlib.org/stable/api/_as_gen/matplotlib.pyplot.semilogx.html), replacing `basex` with `base`  in lines 36 and 46 of file `PySpice/Plot/BodeDiagram.py` fix the problem.

regards,
Mik

cdivry, 2023-10-20:
Hello folks !

didn't get enough rights to create a branch on this rep, in order to make a PR
so here's how to fix the several occurrences of old axes kwargs usage :

[image]

I hope this could be available asap

cdivry, 2023-10-20:
hello, same problem encountered here.

you should have a look on [PR #358 proposal](https://github.com/PySpice-org/PySpice/pull/358)

lionelsaintecluque, 2023-10-23:
I faced the bug and I applied the patch proposed above (PR#358).

It resolves the issue and does not causes unwanted behaviour.

stan3c, 2024-03-30:
This keywords "basex", basey" were replaced in the most recent matplotlib releases.
So you either need to downgrade the matplotlib or fix PySpice.
So really PySpice needs a fix to support latest matplotlib.
see: https://stackoverflow.com/questions/71213873/receiving-typeerror-init-got-an-unexpected-keyword-argument-basey-in

shahnewazRvis, 2024-05-10:
for matplotlib version 3.5.1 
changing function call  axe.semilogx(frequency, phase, base=10, **kwargs) works.

Matplotlib (version 3.5.1) throws error with basex keyword which has been used in BodeDiagram.py