PySpice-org/PySpice #360: analysis=simulator.operating_point() from voltage divider is not working

hebersarmiento, 2023-11-17:
### Environment (OS, Python version, PySpice version, simulator)
Fedora 38 linux, PySpice-1.5 , libngspice-41-1.fc38.x86_64, ngspice-41,python 3.11.

I ran test.py for a voltage divider (code below):

import PySpice
import PySpice.Logging.Logging as Logging
from PySpice.Spice.Netlist import Circuit #Netlist allows us to create a circuit
from PySpice.Unit import *
import sys
logger = Logging.setup_logging()#it will output in the command line

if sys.platform=="linux" or sys.platform=="linux2":
    PySpice.Spice.Simulation.CircuitSimulator.DEFAULT_SIMULATOR = 'ngspice-subprocess'
elif sys.platform=="win32":
    pass

circuit = Circuit('Voltage Divider')

circuit.V('input', 'in', circuit.gnd, 10@u_V)
circuit.R(1, 'in', 'out', 9@u_kΩ)
circuit.R(2, 'out', circuit.gnd, 1@u_kΩ)

print("The circuit/Netlist:\n\n",circuit)

simulator = circuit.simulator(temperature=25, nominal_temperature=25)
print("The simulator:\n\n",simulator)

#next line breaks!
#analysis=simulator.operating_point() 

exit()

If I remove comment from the line analysis= simulator.operating_point() I get the following error:

023-11-17 11:46:08,495 - PySpice.Spice.NgSpice.Server.SpiceServer.__call__ - INFO - Start the spice subprocess
Traceback (most recent call last):
  File "/path/test.py", line 29, in <module>
    analysis=simulator.operating_point()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/.local/lib/python3.11/site-packages/PySpice/Spice/Simulation.py", line 1194, in operating_point
    return self._run('operating_point', *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/.local/lib/python3.11/site-packages/PySpice/Spice/NgSpice/Simulation.py", line 76, in _run
    raw_file = self._spice_server(spice_input=str(self))
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/.local/lib/python3.11/site-packages/PySpice/Spice/NgSpice/Server.py", line 162, in __call__
    return RawFile(stdout, number_of_points)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/.local/lib/python3.11/site-packages/PySpice/Spice/NgSpice/RawFile.py", line 170, in __init__
    raw_data = self._read_header(stdout)
               ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/.local/lib/python3.11/site-packages/PySpice/Spice/NgSpice/RawFile.py", line 192, in _read_header
    self.circuit_name = self._read_header_field_line(header_line_iterator, 'Circuit')
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/.local/lib/python3.11/site-packages/PySpice/Spice/RawFile.py", line 243, in _read_header_field_line
    raise NameError("Expected label %s instead of %s" % (expected_label, label))
NameError: Expected label Circuit instead of Note

Any help is appreciated!

cyber-g, 2023-11-17:
Duplicate of : https://github.com/PySpice-org/PySpice/issues/352

Fixed in : 
https://github.com/PySpice-org/PySpice/pull/353

hebersarmiento, 2023-11-17:
Thank you, I am sorry for now seeing this. Now it is working!

closed 2023-11-17 by hebersarmiento