PySpice-org/PySpice #379: Incorrectly handles stderr outputs that aren't errors from libngspice version 43

mbacvanski, 2024-10-18:
### Environment (OS, Python version, PySpice version, simulator)
* MacOS 15.0
* Python 3.10.15
* PySpice version 1.5
* libngspice version 43

### Issue
I installed `libngspice` via `brew install libngspice`, which gave libngspice version 43. I am trying to run this code:
```python
from PySpice.Spice.Netlist import Circuit
from PySpice.Unit import *

circuit = Circuit('RC Low-Pass Filter')

# Define components and their connections
circuit.V(1, 'input', circuit.gnd, 10@u_V)  # Voltage source
circuit.R(1, 'input', 'output', 1@u_kΩ)      # Resistor
circuit.C(1, 'output', circuit.gnd, 1@u_uF)  # Capacitor

simulator = circuit.simulator(temperature=25, nominal_temperature=25)
analysis = simulator.transient(step_time=1@u_us, end_time=10@u_ms)
```

On running this, I get the error:

```
Warning: can't find the initialization file spinit.
Unsupported Ngspice version 43
Using SPARSE 1.3 as Direct Linear Solver

---------------------------------------------------------------------------
NgSpiceCommandError                       Traceback (most recent call last)
Cell In[2], line 14
     11 circuit.C(1, 'output', circuit.gnd, 1@u_uF)  # Capacitor
     13 simulator = circuit.simulator(temperature=25, nominal_temperature=25)
---> 14 analysis = simulator.transient(step_time=1@u_us, end_time=10@u_ms)

File ~/mambaforge/envs/am/lib/python3.10/site-packages/PySpice/Spice/Simulation.py:1214, in CircuitSimulator.transient(self, *args, **kwargs)
   1213 def transient(self, *args, **kwargs):
-> 1214     return self._run('transient', *args, **kwargs)

File ~/mambaforge/envs/am/lib/python3.10/site-packages/PySpice/Spice/NgSpice/Simulation.py:119, in NgSpiceSharedCircuitSimulator._run(self, analysis_method, *args, **kwargs)
    116 # load circuit and simulation
    117 # Fixme: Error: circuit not parsed.
    118 self._ngspice_shared.load_circuit(str(self))
--> 119 self._ngspice_shared.run()
    120 self._logger.debug(str(self._ngspice_shared.plot_names))
    121 self.reset_analysis()

File ~/mambaforge/envs/am/lib/python3.10/site-packages/PySpice/Spice/NgSpice/Shared.py:1196, in NgSpiceShared.run(self, background)
   1193 #  in the background thread and wait until the simulation is done
   1195 command = 'bg_run' if background else 'run'
-> 1196 self.exec_command(command)
   1198 if background:
   1199     self._is_running = True

File ~/mambaforge/envs/am/lib/python3.10/site-packages/PySpice/Spice/NgSpice/Shared.py:855, in NgSpiceShared.exec_command(self, command, join_lines)
--> 855     raise NgSpiceCommandError("Command '{}' failed".format(command))
    857 if join_lines:
    858     return self.stdout

NgSpiceCommandError: Command 'run' failed
```

It seems that the error is due to the handling of stderr in `Spice/NgSpice/Shared.py` ([around line 625](https://github.com/PySpice-org/PySpice/blob/fe08718d5a1a9af94576b5aef1fa81910f09b037/PySpice/Spice/NgSpice/Shared.py#L625C1-L626C1)). When anything appears in stderr that does not begin with `'Warning:'`, `self._error_in_stderr` is set to True, causing an `NgSpiceCommandError` to be raised [here.](https://github.com/PySpice-org/PySpice/blob/fe08718d5a1a9af94576b5aef1fa81910f09b037/PySpice/Spice/NgSpice/Shared.py#L851) It appears that the latest version of libngspice can output `Using SPARSE 1.3 as Direct Linear Solver` to stderr, which leads to simulator crashing despite no actual error appearing in stderr. 

If I comment out the line 625 setting `self._error_in_stderr = True`, then the simulation completes with no issues.

Kreijstal, 2024-10-28:
ah yes, I have the same issue... is this a ngspice bug?

mbacvanski, 2024-10-28:
I agree it's odd that they print that info statement to stderr. However it looks that PySpice doesn't (want to?) support `libngspice` versions above 34 (according to the last update [here](https://pypi.org/project/PySpice/)), so I'm not sure if this quick fix is a long-term solution. 

I got around the issue by compiling [`libngspice` version 34](https://sourceforge.net/projects/ngspice/files/ng-spice-rework/old-releases/34/) from scratch for my system. It would be nice if this note was included in the documentation :)

tapegoji, 2025-03-05:
1.6 devel version handles them without issues. However, devel version needs some work. let me know if you are still interested in this.