PySpice-org/PySpice #380: Does PySpice run with the iOS app Juno?

Kyhiendo, 2024-10-26:
[image]
### Environment (OS, Python version, PySpice version, simulator)

### Expected Behaviour

PySpice 1.5 seemed to install fine, in an iOS app called Juno

### Actual Behaviour

An example, that ran in Spyder (Anaconda), did not run in Juno.  Error message is as seen in the screen shot attached
[image]

### Steps to reproduce the behaviour

Kreijstal, 2024-11-23:
can you compile ngspice in juno?

Kyhiendo, 2024-11-25:
Hi,Yes, PySpice appear to install ok in Juno.Can I send you a PySpice script that I tried to run in Juno?  You can see the error messages from that.Thanks,Hien Do

Kyhiendo, 2024-11-25:
Hi,Here is a script that I ran.  I copied this from the PySpice example page.  This script runs ok in Anaconda & Spyder on s laptop.I am sure what stderr and stdout are?  I am very new to Python...  I am a hardware engineer who is trying to learn.Thanks,Hien Do***import numpy as npimport matplotlib.pyplot as plt#import PySpice.Logging.Logging as Logging#logger = Logging.setup_logging()from PySpice.Probe.Plot import plotfrom PySpice.Spice.Netlist import Circuitfrom PySpice.Unit import *from scipy.optimize import curve_fitfigure = plt.figure(1, (20, 10))element_types = ('capacitor', 'inductor')for element_type in ('capacitor', 'inductor'):    circuit = Circuit(element_type.title())    # Fixme: compute value    source = circuit.PulseVoltageSource('input', 'in', circuit.gnd,                          ***@***.***_V, ***@***.***_V,                          ***@***.***_ms, ***@***.***_ms)    circuit.R(1, 'in', 'out', ***@***.***_kΩ)    if element_type == 'capacitor':        element = circuit.C        value = ***@***.***_uF        # tau = RC = 1 ms    else:        element = circuit.L        # Fixme: force component value to an Unit instance ?        value = ***@***.***_H        # tau = L/R = 1 ms    element(1, 'out', circuit.gnd, value)    # circuit.R(2, 'out', circuit.gnd, kilo(1)) # for debug    if element_type == 'capacitor':        tau = circuit['R1'].resistance * circuit['C1'].capacitance    else:        tau = circuit['L1'].inductance / circuit['R1'].resistance    simulator = circuit.simulator(temperature=25, nominal_temperature=25)    step_time = ***@***.***_us    analysis = simulator.transient(step_time=step_time, end_time=source.period*3)    # Let define the theoretical output voltage.    if element_type == 'capacitor':        def out_voltage(t, tau):            # Fixme: TypeError: only length-1 arrays can be converted to Python scalars            return float(source.pulsed_value) * (1 -  np.exp(-t / tau))    else:        def out_voltage(t, tau):            return float(source.pulsed_value) * np.exp(-t / tau)    # Fixme: get step_time from analysis    # At t = 5 tau, each circuit has nearly reached it steady state.    i_max = int(5 * tau / float(step_time))    popt, pcov = curve_fit(out_voltage, analysis.out.abscissa[:i_max], analysis.out[:i_max])    tau_measured = popt[0]    # Fixme: use Unit().canonise()    print('tau {0} = {1}'.format(element_type, tau.canonise().str_space()))    print('tau measured {0} = {1:.1f} ms'.format(element_type, tau_measured * 1000))    if element_type == 'capacitor':        axe = plt.subplot(121)        title = "Capacitor: voltage is constant"    else:        axe = plt.subplot(122)        title = "Inductor: current is constant"    axe.set_title(title)    axe.grid()    current_scale = 1000    plot(analysis['in'])    plot(analysis['out'])    # Fixme: resistor current, scale    plot(((analysis['in'] - analysis.out)/circuit['R1'].resistance) * current_scale)    axe.axvline(x=float(tau), color='red')    axe.set_ylim(-11, 11)    axe.set_xlabel('t [s]')    axe.set_ylabel('[V]')    axe.legend(('Vin [V]', 'Vout [V]', 'I'), loc=(.8,.8))       plt.tight_layout()    plt.show()

sure, but can you send the stderr and stdout

Kreijstal, 2024-11-25:
I think you have to send the file as attachment, not in the email