PySpice-org/PySpice #383: SubCircuit nodes bug

NobilisManius, 2024-11-21:
### Environment (Windows, Python 3.12, PySpice 1.5, None)
Hello. I am parsing some .sp files to check SubCircuit nodes, but I encountered what seems to be a positioning bug. Could someone help?

```
.subckt CELL vdd gnd i_source=0
icell vdd gnd i_source
.ends

...

x_J1       vdd_2       gnd_4    CELL  i_source=2.3
x_J2       vdd_3       gnd_7    CELL  i_source=4.3
x_J3       vdd_10      gnd_5    CELL  i_source=6.5
x_J4       vdd_11      gnd_6    CELL  i_source=9.1
```

### Expected Behaviour

```
'X_J1' = {SubCircuitElement} SubCircuitElement X_J1
 node_names = {list: 2} ['vdd_2', 'gnd_4']
 nodes = {list: 2} [Node vdd_2, Node gnd_4]
 subcircuit_name = {str} 'CELL'
 parameters = {dict: 1} {'i_source': 2.3}
```

### Actual Behaviour
Node 1 is node 2. Node 2 is param. Name is node 2.

```
'X_J1' = {SubCircuitElement} SubCircuitElement X_J1
 * = {Pin} Pin None of X_J1 on node gnd_4
 ALIAS = {str} 'X'
 PINS = {list: 1} [<PySpice.Spice.Netlist.PinDefinition object at 0x0000029C7F86F770>]
 PREFIX = {str} 'X'
 enabled = {bool} True
 name = {str} 'X_J1'
 node_names = {list: 2} ['gnd_4', '2.3']
 nodes = {list: 2} [Node gnd_4, Node 2.3]
 parameters = {dict: 0} {}
 pins = {list: 2} [Pin None of X_J1 on node gnd_4, Pin None of X_J1 on node 2.3]
 raw_spice = {str} ''
 subcircuit_name = {str} 'vdd_2'
```
```
'X_J2' = {SubCircuitElement} SubCircuitElement X_J2
 * = {Pin} Pin None of X_J2 on node gnd_7
 ALIAS = {str} 'X'
 PINS = {list: 1} [<PySpice.Spice.Netlist.PinDefinition object at 0x0000029C7F86F770>]
 PREFIX = {str} 'X'
 enabled = {bool} True
 name = {str} 'X_J2'
 node_names = {list: 2} ['gnd_7', '4.3']
 nodes = {list: 2} [Node gnd_7, Node 4.3]
 parameters = {dict: 0} {}
 pins = {list: 2} [Pin None of X_J2 on node gnd_7, Pin None of X_J2 on node 4.3]
 raw_spice = {str} ''
 subcircuit_name = {str} 'vdd_3'
```

Kreijstal, 2024-11-23:
can you upload the sp files?

NobilisManius, 2024-11-25:
> can you upload the sp files?

Oh, sure:
[test.txt](https://github.com/user-attachments/files/17899216/test.txt)

Kreijstal, 2024-11-25:
> > can you upload the sp files?
> 
> Oh, sure: [test.txt](https://github.com/user-attachments/files/17899216/test.txt)

how are you parsing it in pyspice?

NobilisManius, 2024-11-25:
> > > can you upload the sp files?
> > 
> > 
> > Oh, sure: [test.txt](https://github.com/user-attachments/files/17899216/test.txt)
> 
> how are you parsing it in pyspice?

```
filename = 'data/test.sp'

parsed_data = SpiceParser(filename)
cir = parsed_data.build_circuit()
```

tapegoji, 2025-03-05:
There is a new parser ( written from scratch in 1.6) try it out.