PySpice-org/PySpice #388: Include from inside SubCircuitFactory

mbacvanski, 2024-12-04:
I have a few nested subcircuits, which I'm building with SubCircuitFactory. It seems that when I use a component in an inner subcircuit, it needs to be included in its immediately containing subcircuit, which means I need to be able to have an `.include` statement from SubCircuitFactory. However, it seems like I can only add `.include` statements from `Circuit`s. How can I get around this?

As an example, this is what I need

```
.subckt A input output
.include opamp.sub
.subckt B input output
* inside here I use the opamp
.ends B
.ends A
.end
```

but there is no `self.include(...)` method from `SubCircuitFactory`, so it seems like I can only do this, which doesn't work:

```
.include opamp.sub
.subckt A input output
.subckt B input output
* inside here I use the opamp
.ends B
.ends A
.end
```

Kreijstal, 2025-02-02:
ahh yes I think I have a very similar problem

tapegoji, 2025-04-18:
@mbacvanski This is fixed now if you are still intersted.