PySpice-org/PySpice #394: spice parser fails on model imports.

tapegoji, 2025-02-28:
I have been trying to fix it but seems that there is a problem at         data = ElementData.elements[self._letter]

line 316 in HighLeverParser.py. 

ElementData.elements dict is empty. seems to be an import issue. 

please help!. this might be a quick check for anyone who know the code. 

Thanks

Kreijstal, 2025-02-28:
write a MWE

tapegoji, 2025-02-28:
I am using 1.6 devel branch. I am simply trying to run buck converter example in the switched mode power supplies. 

PySpice/examples/switched-power-supplies/buck-converter.py 

What I have found is the problem with importing ElementParameterMetaClass. when importing it, all the _classes are empty that is why ElementData.elements are empty. 

This is not a problem with this example, it's a bigger problem with the devel code for parsing the library. I am scratching my head here and not sure what is happening. 

Also, another note: Try to run ElementData.py on it's own. it can find all the elements and prints them out. 

My take, if you import ElementData from another file. It tries to ElementParameterMetaClass from Element.py. then Element tries to import ElementData which ElementData is importing ElementParameterMetaClass. I think there is an infinite loop here which python is trying to avoid. therefore it simply breaks the loop without executing it.

Kreijstal, 2025-02-28:
ah yeah devel code doesn't work. That's why I just improve on top of the 1.5 version, the devel branch is broken afaik...
If you manage to fix it... do let us know!!

tapegoji, 2025-02-28:
I was able to run a few tests with devel branch when I switched to python 3.12. It works well with ngspice 44 and did not see any issue with that. 
I just can't manage to figure out the spice parser issue.

tapegoji, 2025-02-28:
I made a temp fix. Look to see what do you think. You might have a better idea of how to fix it. 

[https://github.com/tapegoji/PySpice/tree/elementdata_fix](url)

closed 2025-04-18 by tapegoji