PySpice-org/PySpice #399: Ngspice .op : Non Fatal Notes and diagnosis written in stderr treated as errors

xvan, 2025-05-12:
PySpice - 1.5
NgSpice - 36 (Unsupported, I know,  but couldn't find easy way to upgrade it in ubuntu)

ngspice can execute the .op, but outputs some notes:

```
Note: Starting dynamic gmin stepping
Trying gmin =   1.0000E-03 Note: One successful gmin step
[...]
Note: Dynamic gmin stepping completed
```
PySpice can't handle them and aborts. Workaround is to ignore them

```
--- lib/python3.10/site-packages/PySpice/Spice/NgSpice/Shared.py       2025-05-12 19:17:17.906873929 -0300
+++ lib/python3.10/site-packages/PySpice/Spice/NgSpice/Shared2.py      2025-05-12 19:20:50.727956067 -0300

@@ -618,7 +618,8 @@
         prefix, _, content = message.partition(' ')
         if prefix == 'stderr':
             self._stderr.append(content)
-            if content.startswith('Warning:'):
+            if content.startswith('Warning:') or content.startswith('Note:') or content.startswith('Trying'):
                 func = self._logger.warning
             # elif content.startswith('Warning:'):
             else:
```

Netlist with Issues: OPA2735_ng.net

```
v1 c1 0 3.7
v2 c2 c1 3.7
v3 c3 c2 3.7
v4 c4 c3 3.6
r1 out_op1 in1m    2.700000000000000e+04   
r2 in1m n001    2.200000000000000e+03   
r3 in1p n003    2.200000000000000e+03   
r4 in1p 0    2.700000000000000e+04   
v5 vcc 0 6
v6 0 vdd 6
r5 n001 inmuxon    9.000000000000000e+01   
r6 n003 n002    9.000000000000000e+01   
r7 0 inmuxon    3.000000000000000e+03   
r8 0 n002    3.000000000000000e+03   
r9 c4 n002    1.100000000000000e+04   
r10 c3 inmuxon    1.100000000000000e+04   
q.xu1.q21 xu1.7 xu1.8 xu1.9 xu1:qnl 
r.xu1.r77 xu1.10 xu1.11 2 
r.xu1.r78 xu1.12 xu1.11 2 
r.xu1.r79 xu1.13 xu1.8 100 
r.xu1.r80 xu1.14 xu1.15 100 
r.xu1.r81 xu1.16 vcc 7 
r.xu1.r82 vdd xu1.17 7 
r.xu1.r84 xu1.18 xu1.19 700 
r.xu1.r85 xu1.20 xu1.21 1 
r.xu1.r86 xu1.9 xu1.22 1 
d.xu1.d21 out_op1 vcc xu1:dd 
d.xu1.d22 vdd out_op1 xu1:dd 
d.xu1.d23 xu1.23 0 xu1:din 
d.xu1.d24 xu1.24 0 xu1:din 
i.xu1.i24 0 xu1.23 0.1e-3
i.xu1.i25 0 xu1.24 0.1e-3
e.xu1.e25 xu1.9 0 vdd 0 1
e.xu1.e26 xu1.21 0 vcc 0 1
d.xu1.d25 xu1.25 0 xu1:dvn 
d.xu1.d26 xu1.26 0 xu1:dvn 
i.xu1.i26 0 xu1.25 0.1e-3
i.xu1.i27 0 xu1.26 0.1e-3
e.xu1.e27 xu1.27 in1m xu1.25 xu1.26 0.1
[... 468 lines cut ...]
2025-05-12 19:17:36,249 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Trying gmin =   1.0000E-03 Note: One successful gmin step
2025-05-12 19:17:36,249 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Trying gmin =   1.0000E-04 Note: One successful gmin step
2025-05-12 19:17:36,250 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Trying gmin =   1.0000E-05 Note: One successful gmin step
2025-05-12 19:17:36,250 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Trying gmin =   1.0000E-06 Note: One successful gmin step
2025-05-12 19:17:36,250 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Trying gmin =   1.0000E-07 Note: One successful gmin step
2025-05-12 19:17:36,250 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Trying gmin =   1.0000E-08 Note: One successful gmin step
2025-05-12 19:17:36,251 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Trying gmin =   1.0000E-09 Note: One successful gmin step
2025-05-12 19:17:36,251 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Trying gmin =   1.0000E-10 Note: One successful gmin step
2025-05-12 19:17:36,251 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Trying gmin =   1.0000E-11 Note: One successful gmin step
2025-05-12 19:17:36,251 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Trying gmin =   1.0000E-12 Note: One successful gmin step
2025-05-12 19:17:36,251 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Trying gmin =   1.0000E-12 Note: One successful gmin step
2025-05-12 19:17:36,252 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Note: Dynamic gmin stepping completed
Traceback (most recent call last):
  File "/home/jzuloaga/rcgapy/demo_fail.py", line 23, in <module>
    analysis = simulator.operating_point()
  File "/home/jzuloaga/rcgapy/.venv/lib/python3.10/site-packages/PySpice/Spice/Simulation.py", line 1194, in operating_point
    return self._run('operating_point', *args, **kwargs)
  File "/home/jzuloaga/rcgapy/.venv/lib/python3.10/site-packages/PySpice/Spice/NgSpice/Simulation.py", line 119, in _run
    self._ngspice_shared.run()
  File "/home/jzuloaga/rcgapy/.venv/lib/python3.10/site-packages/PySpice/Spice/NgSpice/Shared.py", line 1193, in run
    self.exec_command(command)
  File "/home/jzuloaga/rcgapy/.venv/lib/python3.10/site-packages/PySpice/Spice/NgSpice/Shared.py", line 852, in exec_command
    raise NgSpiceCommandError("Command '{}' failed".format(command))
PySpice.Spice.NgSpice.Shared.NgSpiceCommandError: Command 'run' failed
```

xvan, 2025-05-12:
Found similar bug

closed 2025-05-12 by xvan

Kreijstal, 2025-05-13:
yep, welcome to the club

xvan, 2025-05-13:
> @xvan <https://github.com/xvan> I may have fixed this already in InSpice.
> check it out.

What's InSpice?

Kreijstal, 2025-05-13:
tapegojis fork

tapegoji, 2025-05-13:
> What's InSpice?

https://github.com/Innovoltive/InSpice