PySpice-org/PySpice #400: B source with current_expression dependent on another component current

GoncaloAguiarN3O, 2025-05-13:
Hi all,

I am trying to define a constant power load in PySpice by using behavioral source (B) with a current expression that should characterize the current like this: `i = power / voltage`

In PySpice this would look something like this:

`circuit.B(1,'LOAD',circuit.gnd,i = _current_expression_)
`
The problem is that I cannot find a way in PySpice to call a node voltage (like v(node)) in the _current_expression_. In LTSpice, for example, if a node is called "LOAD", I can simply write the expression as I=POWER/V(LOAD).

Does anyone know how to solve this problem?

Thank you in advance!

tapegoji, 2025-05-13:
use raw_spice keyword when defining B. That way you can define it the way you want it.

GoncaloAguiarN3O, 2025-05-14:
Thank @tapegoji, that works.

Here's the final test code:

```
circuit = Circuit('Constant power load')

voltage = 10
power =25

circuit.V(1, 'inp', circuit.gnd, voltage)
raw_spice_expression = f"i = {power}/V(inp)"
circuit.B(1, 'inp', circuit.gnd, raw_spice = raw_spice_expression)
```

closed 2025-05-14 by GoncaloAguiarN3O

tapegoji, 2025-05-14:
@GoncaloAguiarN3O 

If you face other issues, go and check out my fork which has more updates and a lot of problems has been solved. https://github.com/Innovoltive/InSpice